bblanchon/ArduinoJson #258: Adding objects in a loop as array elements to JSON

mweisshaupt1988, 2016-04-05:
Hi, I need a json library for an embedded plattform which is not as restricted as an Arduino but which also can't use gigabytes of memory. So I came across your library and I wanted to do some basic testing on my PC first.

The library and documentation is very good but I have one problem that I couldn't solve until now.
We will need to create JSONs not only for simple key-value objects but also for a much larger object with up to 50 sub objects. These objects contain messages, timestamps  and some other stuff.

I read in the documentation that the library does not automatically copy the values and that they need to be present in memory until the final string has been printed. This is good for memory usage but bad for my loop :D

I have attached my testcode where I add various items to the json objects and I have problems with the messages array. I need to copy data into a json array with up to 50 elements. The items need to be stored somewhere so I created a two dimensional array for the serialized value. This works for now.

But then I can't figure out how to  get the message objects into the main object. The wiki mentions this function: `ArduinoJson::Internals::Unparsed` but using it to assign the value doesn't compile because this line prevents it from doing so:
`typename TypeTraits::EnableIf<JsonObject::CanSet<T>::value,                        JsonObjectSubscript<TKey> >::type&`

I have two questions:
1. Is there a better way to add objects as array elements to the json string in a loop?
2. Is the function `ArduinoJson::Internals::Unparsed` supported on x86 machines (or at all)? I couldn't get it to work with Visual Studio 2015 on Windows 10 as well as GCC 4.8.2 on Ubuntu 14.04

Here is my testcode:
```cpp
#include <stdlib.h>
#include <ArduinoJson.h>

int main()
{
    //StaticJsonBuffer<40000> jsonBuffer;
    DynamicJsonBuffer jsonBuffer(10000);
    char charBuffer[40000];

    JsonObject& rootObject = jsonBuffer.createObject();
    rootObject["Time"] = "13:37";
    rootObject["Date"] = "01.01.20016";

    JsonObject& dataObject = rootObject.createNestedObject("data");
    dataObject["FullName"] = "A simple Test project";
    dataObject["ProjectName"] = "Simple";
    dataObject["Version"] = "1.0.2";

    JsonArray& moredata = dataObject.createNestedArray("counter");
    for (int i = 0; i < 500; i++)
    {
        moredata.add(i);
    }

    JsonObject& permissionsObject = rootObject.createNestedObject("permissions");
    permissionsObject["readable"] = true;
    permissionsObject["writable"] = true;
    permissionsObject["executable"] = false;

    //JsonArray& messagesObject = rootObject.createNestedArray("messages");
    char bufferArray[10][4000];
    for (int i = 0; i < 10; i++)
    {
        StaticJsonBuffer<4000> loopBuffer;

        JsonObject& loopObject = loopBuffer.createObject();
        loopObject["Message"] = "Some text here...";
        loopObject["Timestamp"] = 1234567890;
        loopObject["Var1"] = "Some";
        loopObject["Var2"] = "other";
        loopObject["Var3"] = "stuff";
        loopObject["Var4"] = "here";
        loopObject["Var5"] = "...";

        loopObject.printTo(bufferArray[i], 4000);
        //messagesObject.add(ArduinoJson::Internals::Unparsed(bufferArray[i])); // <!-- not accepted
    }

    // Trying a different approach...
    char subObject[10 * 4000];
    snprintf(subObject, 10 * 4000, "[%s, %s, %s, %s, %s, %s, %s, %s, %s, %s]",
        bufferArray[0],
        bufferArray[1],
        bufferArray[2],
        bufferArray[3],
        bufferArray[4],
        bufferArray[5],
        bufferArray[6],
        bufferArray[7],
        bufferArray[8],
        bufferArray[9]
    );

    rootObject["messages"] = ArduinoJson::Internals::Unparsed(subObject); // <!-- not accepted

    // Copied from the wiki but also not working...
    rootObject["value2"] = ArduinoJson::Internals::Unparsed("[[\"Yes\",\"No\"],[\"Maybe\"]"); // <!-- not accepted

    rootObject.prettyPrintTo(charBuffer, sizeof(charBuffer));
    FILE* file = fopen("testfile.json", "w");
    fprintf(file, "%s", charBuffer);
    fclose(file);

    return EXIT_SUCCESS;
}
```

bblanchon, 2016-04-05:
Hi,

I think you are looking for `JsonBuffer::strdup()`.
It creates a copy of the string in the `JsonBuffer`.

Thanks for pointing out that the `ArduinoJson::Internals::Unparsed` trick doesn't work, I'll open an issue for that.

bblanchon, 2016-04-05:
I updated [Avoiding pitfalls: 6. Do not assume that strings are copied](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#6-do-not-assume-that-strings-are-copied) in the wiki because it didn't mentioned `JsonBuffer::strdup()`
Sorry about that.

mweisshaupt1988, 2016-04-05:
Wow, I didn't expect an answer that fast, great! :smiley: 

I guess the line `jsonBuffer.strlen(buffer)` in the array example should be `jsonBuffer.strdup(buffer)` but I don't know if I should modify your wiki...

I didn't know that `JsonBuffer::strdup()` could be used for that. Dublicating the serialized object (see below) into the array copies the string but then I have escapes everywhere in the Array so this does not work out for me...
`messagesObject.add(jsonBuffer.strdup(bufferArray[i]));`

I think as soon as #259 is fixed, it should be the best way to do this in combination with `JsonBuffer::strdup()` or am I mistaken?

Thank you for your help.

bblanchon, 2016-04-05:
Well, the goal is to avoid the handmade serialization with `snprintf()` and get rid of the `bufferArray`.

Something like:

```c++
JsonArray& messages = rootObject.createNestedArray("messages");
for (int i = 0; i < 10; i++) {
    JsonObject& msg = messages.createNestedObject();
    msg["Timestamp"] = 1234567890;
    msg["Message"] = "Some string literal";
    msg["Var1"] = jsonBuffer.strdup(someStringVariable);
    // ...
}
```

Please edit the wiki, I leave the typo so you can fix it :-)

mweisshaupt1988, 2016-04-05:
I fixed the typo...
This solution if perfect and also much cleaner. I also wouldn't like the `snprintf()` approach in production code.

Thx a lot!

closed 2016-04-05 by mweisshaupt1988

bblanchon, 2016-04-05:
You're welcome.
Thank you for using ArduinoJson.

jmohin, 2017-09-30:
Can you explain why this gives the output it does? I'm using the library on a Particle Core v1.

 ```
   uint8_t buffer[8][12]={0};
    for( int row = 0; row < 8; row++ ) {
      for( int col = 0; col < 12; col++ ) {
          buffer[row][col]=100;
      }
    }

    DynamicJsonBuffer jsonBuffer;
    JsonObject& root = jsonBuffer.createObject();

    for (int row = 0; row < 8; row=row+1) {
        JsonArray& baVals = root.createNestedArray(String(row).c_str());
        for(int col=0; col<12; col=col+1){
           baVals.add(buffer[row][col]);
        }
    }
    root.printTo(Serial);
```

This code outputs:
{"0":[100,100,100,100,100,100,100,100,100,100,100,100],"2":[100,100,100,100,100,100,100,100,100,100,100,100],"5":[100,100,100,100,100,100,100,100,100,100,100,100],"7":[100,100,100,100,100,100,100,100,100,100,100,100]}

Why is it skipping loop iterations?  I suspect it has something to do with me using the same JsonArray "baVals" and that the memory is getting overwritten, but I truly don't see a better way to do this without some horrendously inefficient code (loops are bad enough).