bblanchon/ArduinoJson #260: String data in variable.

MYeager1967, 2016-04-07:


bblanchon, 2016-04-07:
Hi Mike,

This is fixed since [v5.0](https://blog.benoitblanchon.fr/arduino-json-v5-0/).
The most likely explanation is that you're using an older version.

MYeager1967, 2016-04-07:
I just downloaded the new version from Github yesterday. Had to figure out how to change a few things over from the way they were done before (or at least the way I was doing them). Still have a few things to figure out. Is it possible that parts of the old library were cached somewhere and still being used? I had renamed the old library but now I’ve removed it completely. I’m going to run a few more tests to see if it works. If not, I can post the routine in question if you like.

 

Great library!!!

MYeager1967, 2016-04-07:
After a LOT of fiddling and upgrading. It seems to be working properly now.

bblanchon, 2016-04-07:
Great!
I'm closing the issue then.

closed 2016-04-07 by bblanchon