bblanchon/ArduinoJson #261: Create a JSON object inside a JSON object

mgarcia96, 2016-04-11:
Hey, can I create something like this?

`
{
 "hello": 1,
 "data": {
  "value": 123
 }
}
`

hemantsangwan, 2016-04-12:
Kindly follow [Encoding JSON](https://github.com/bblanchon/ArduinoJson/wiki/Encoding%20JSON#objects) and [Bag of Tricks](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks)

bblanchon, 2016-04-12:
Hi Marc,

Here is how to do this:

```c++
JsonObject& root = jsonBuffer.createObject();
root["hello"] = 1;
JsonObject& data = root.createNestedObject("data");
data["value"] = 123;
```

As @hemantsangwan said, you should be able to find all the required material [in the wiki](https://github.com/bblanchon/ArduinoJson/wiki)

mgarcia96, 2016-04-12:
Great! I only saw the createNestedArray so it didn't work.

Thanks!

closed 2016-04-12 by mgarcia96

bblanchon, 2016-04-12:
You're welcome Marc, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: