bblanchon/ArduinoJson #262: Available Memory For Parsing JSON Data

pieman64, 2016-04-18:
@bblanchon liking your work Benoît.

I have looked through the WiKi regarding memory allocation with your library but I still have a few questions.

We have a json 'string' to read with our 1M ESP8266 (64K SPIFFS) and the size of the 'string' is approximately 5400 characters. Not sure string is the correct term, object maybe, but either way it is the full payload we receive from a website. Our ESP crashes if we set MAX_CONTENT_SIZE to 5500. It seems ok at 2500 but this is not enough for the full HTTP response.

Is there anyway of us parsing a 5500 character http response with our ESP?
We only require a small part of the response but having very limited knowledge of json and parsing I'm not sure if it is possible to just store the small parts we require. We know the structure of the http response if that helps.

bblanchon, 2016-04-18:
Hi @pieman64,

I assume that you are talking about the "JsonHttpClient" example.

I believe that your ESP8266 is crashing because of a stack overflow.
Indeed, the stack is set to 4KB by default on this device, so 5500 bytes is way too much.

The solution is to move the `response` to the heap.

Instead of:

```c++
char response[MAX_CONTENT_SIZE];
// ...
```

do:

```c++
char* response = malloc(MAX_CONTENT_SIZE);
// ...
free(response);
```

(you can use a `std::unique_ptr<char[]>` or a `std::vector<char>` if you prefer).

Also, you should use a `DynamicJsonBuffer` instead of a `StaticJsonBuffer`.

Please tell me if it works.

pieman64, 2016-04-18:
Thanks Benoît I will look at heap and DynamicJsonBuffer. Just waiting for my machine to upgrade the ESP8266 boards to 2.2.0. I was thinking Dynamic was only used if I didn't know what response the http server was sending.

bblanchon, 2016-04-18:
I added a section in the FAQ:
[Why does my device crash or reboot?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-does-my-device-crash-or-reboot)

pieman64, 2016-04-18:
Thanks I'll study the updated FAQ.

bblanchon, 2016-04-24:
Did it work as expected?

closed 2016-04-27 by bblanchon