bblanchon/ArduinoJson #263: Can't get to my variables outside the parsing function

viktak, 2016-04-20:
Hi,

This is more like a question rather than a bug, and most likely stems from me a c++ beginner.
Writing ESP8266 code, I have a variable declared as:
`const char* ssid;`
Then, in the function where I load some configuration data, I have the following code:
`data.ssid = root["ssid"];`
And it is working just fine, I can verify it right after, still in the function, by:
`Serial.println(ssid);`
However, when I want to access `ssid` from anywhere else in my program (`Serial.println(ssid);`) I get garbage.
I wonder what it the (best) way to access this variable throughout my program.

Thanks for the help in advance.

bblanchon, 2016-04-20:
Hi @viktak,

This is a very common beginner mistake in C.

Here is the canonical example:

```c++
const char* getString()
{
    char str[] = "hello world";
    return str;
}
```

If you try to use later the return value of `getString()`, you're very likely to get garbage.
This is because `getString()` returns a pointer to a temporary variable.
The same thing probably happens in your code.

To work around this problem, you need to make a copy of the string.
You can do that "manually" with `strcpy()`, or "implicitly" with the `String` class:

```c++
void getString(char* destination) {
   strcpy(destination, "hello world");
}

String getString() {
    char str[] = "hello world";
    return str; // <- implicit construction of a String
}
```

Note that in the case of ArduinoJson, you are returning a pointer to a location in the JSON input string.
So you can avoid making a copy if you can guarantee that this variable is still in memory at the time you use the pointer.

Regards,

sticilface, 2016-04-20:
This is a bit of a tricky concept to get to grips with in c, that certainly baffled me a lot when I started. In a nut shell....

When you declare a const chat * you are really creating a variable which contains an address, a 32bit pointer on esp8266 that says the string, in this case ssid, starts at this location in memory.  The key thing about this location is that it doesn't have to be permanent, it can be temporary. So even if ssid is a global variable the address it points to is temporary as it is in the json buffer. This buffer is deleted the moment the function it's in exits.  This is why it turns to garbage. 

So how to address this. 
1)
Declare ssid as an array of your maximum length, and then copy the data from one to the other. 2) allocate the memory for ssid using new/malloc which means that it is copied there and does not vanish when the function exits. 

I can't give u an example right now as I'm on my phone but can later. Hope that helps.

viktak, 2016-04-20:
Thank you guys, let me try this, and will report back.

While we are at it: How do you mark an issue here as question, rather than bug? I see some people doing it, but I couldn't mark my thread as such, although I was 99% it was not a bug.

thanks again!

bblanchon, 2016-04-20:
@viktak, I think only repository administrators can do that.

viktak, 2016-04-20:
> I think only repository administrators can do that.

@bblanchon  I see, thanks

viktak, 2016-04-20:
@bblanchon , @sticilface : Thank you for the quick response. I immediately managed to get it to work once I saw your explanations.

closed 2016-04-20 by viktak

bblanchon, 2016-04-20:
You're welcome @viktak, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

viktak, 2016-04-20:
@bblanchon  Starred with no hesitation. Your library makes my life a lot easier. It's just my c/c++ skills need some developing.
Once my project is finished, I'll be back to post it on the page you showed.