bblanchon/ArduinoJson #264: Complex Nested Json Objects

bepursuant, 2016-04-21:
Sorry to ask a question you've already answered... but I've been unable to figure out exactly how to implement the solution you mentioned in issue #252 .

I would like to output my json in the following format:
```
{"networks":{
  [0] : {
    "ssid":"asdf",
    "rssi":"test",
    etc etc etc
  },
  [1] : {
    "ssid":jkl",
    "rssi":"fubar",
    etc etc etc
  }
}}
```

I've tried your example in about as many ways as I can think of, but obviously I'm missing something?

```
String scan_network() {
  DEBUG_PRINT(F("Scanning available networks..."));

  WiFi.mode(WIFI_STA);
  WiFi.disconnect();
  byte n = WiFi.scanNetworks();

  DEBUG_PRINT(F("found "));
  DEBUG_PRINT(n);
  DEBUG_PRINT(F("..."));

  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  String retJson;

  for(int i=0;i<n;i++) {
    root["ssid"] = WiFi.SSID(i);
    root["encryption"] = WiFi.encryptionType(i);
    root["rssi"] = WiFi.RSSI(i);
    root["bssid"] = WiFi.BSSIDstr(i);
    root["channel"] = WiFi.channel(i);
    root["isHidden"] = WiFi.isHidden(i);
  }
  root.printTo(retJson);
  DEBUG_PRINTLN(F("ok!"));

  return retJson;
}
```

bblanchon, 2016-04-21:
Hi @bepursuant,

There are three problems in your program:

1. You try to generate invalid JSON
2. The `JsonBuffer` is too small
3. The loop overrides the root values

To address problem 1, I'll simply assume that the JSON is the following:

```json
{
  "networks": [
    {
      "ssid": "foo",
      "rssi": -70
    },
    {
      "ssid": "bar",
      "rssi": -80
    }
  ]
}
``` 

To address problem 2, without further information, I suggest to switch to a `DynamicJsonBuffer`.

To address problem 3, you just need to call `createNestedObject()` in the loop.

Here is the final program:

```c++
String scan_network() {
   // ...
 
  DynamicJsonBuffer jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();

  JsonArray& networks = root.createNestedArray("networks");
  
  for(int i=0;i<n;i++) {
    JsonObject& network = networks.createNestedObject();
    network["ssid"] = WiFi.SSID(i);
    network["encryption"] = WiFi.encryptionType(i);
    network["rssi"] = WiFi.RSSI(i);
    network["bssid"] = WiFi.BSSIDstr(i);
    network["channel"] = WiFi.channel(i);
    network["isHidden"] = WiFi.isHidden(i);
  }

  String retJson;
  root.printTo(retJson);
  return retJson;
}
```

closed 2016-04-27 by bblanchon

bepursuant, 2016-04-28:
This is beautiful! I think a lot of my trouble came from my incorrect JSON requirements. Using the method you outlined above, I was able to convert my entire API to return JSON using this library.

Thank you so much for the thorough answer and detailed response. Is there a reason (I didn't see) DynamicJsonBuffer in the example?

Thanks again! I'm sure this answer will be used by more than just me :)

bblanchon, 2016-04-28:
I chose `DynamicJsonBuffer` because I assumed you were running on an ESP8266.
The stack on the ESP8266 is limited to 4KB, so it's not possible to have a big `StaticJsonBuffer`.

You can optimize the performance of the program by specifying an initial size to the constructor of the `DynamicJsonBuffer`.

I added a link to this question in the FAQ: [How do I create complex nested objects?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-do-i-create-complex-nested-objects)

terryjmyers, 2017-04-16:
I'm having difficulty trying to figure out how to decode double nested objects.  For example I'm using WeatherUnder.com API json calls and sometimes I'll get an error Json:

`{
	"response": 
	{
		"version":"0.1",
		"termsofService":"http://www.wunderground.com/weather/api/d/terms.html",
		"features": {
			"conditions": 1
		},
		"error": 
		{
			"type": "querynotfound"
			,"description": "No cities match your search query"
		}
	}
}`

I'm trying to get at the error description to display message to the user.

Here is my (simplified code):
`	DynamicJsonBuffer jsonBuffer(4000);

	//Load data into Json object and parse
	JsonObject& json = jsonBuffer.parseObject(payload);

if (json.success()) {//success
		if (!json.containsKey(F("current_observation"))) {
			WU.hasError = true;
			return;
		}
WU.temp_f = json[F("current_observation")][temp_f];
}
`

But what I really want is something like this (this of course does not work at all):
`
if (!json.containsKey("response","error")) {
	errortext = json[F("response")]["error"]["description"];
	return;
}`

So how can I a. check if a deeply nested key exists, and b.  get a really deeply nested value?

bblanchon, 2017-04-17:
Hi Terry,

I would do this:

```c++
const char* error = json["response"]["error"]["description"];

if (error)
{
    Serial.println(error);
    return;
}
```

Regards,
Benoit

PS: see [GitHub Syntax Highlighting](https://guides.github.com/features/mastering-markdown/)

terryjmyers, 2017-04-19:
Benoit -
Thanks that worked perfectly.  I guess .containsKey is not able to resolve
[...][...][...][etc].  Would you recommend using that method to determine
if a key worked instead of using .containsKey?
I think I'd recommend updating the wiki with that example to show people
how to extract data from a deeply tested Json, as well as checking to see
if it exists

bblanchon, 2017-04-20:
Here you go: [JsonObject::containsKey()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/containskey/); with an example that should be familiar 😉

terryjmyers, 2017-04-21:
That's fantastic.  I'm glad I can help make your library documentation that
much better.  Its already pretty comprehensive :)

umerfarooq786, 2018-07-21:
{"data":{"id":82,"brand_name":"6s","item_name":"iPhone","fault":"screen","technician_id":1,"dateOfBooking":"2018-07-26","timeOfBooking":"08:06:15","created_at":"2018-07-21 17:04:28","updated_at":"2018-07-21 17:04:28"}}
how to convert into gson??

bblanchon, 2018-07-22:
@umerfarooq786, try the [ArduinoJson Assistant](https://arduinojson.org/v5/assistant/?utm_source=github&utm_medium=issues).

[image]