bblanchon/ArduinoJson #265: decode/unquote raw json string literals

proppy, 2016-04-26:
Is there a reason the function for "parsing" (or unquote) json strings literals are private?
https://github.com/bblanchon/ArduinoJson/blob/master/src/Internals/JsonParser.cpp#L146

In addition to documented method for decoding objects and arrays, I found out other literals could be decode with `Arduino::JsonVariant` or `ArduinoJson::Internals::parse`, but it doesn't seems to be possible to decode raw json string literals with the current public API.

(suggested tag: question)

bblanchon, 2016-04-27:
If I'm not mistaken, you want to be able to do this:

```c++
const char* str = jsonBuffer.parseString(json);
```

and more generally this:

```c++
JsonVariant variant = jsonBuffer.parse(json);
```

I can add this to the library but first, can you please explain why you need this?
I'm worried that you could be using the library incorrectly...

proppy, 2016-04-27:
> ```
> JsonVariant variant = jsonBuffer.parse(json);
> ```

That'd be perfect

> I can add this to the library but first,

I would happily send a PR for this, if you agree that's a good idea to add it.

> I can add this to the library but first, can you please explain why you need this?

The reason I need this, is that I'm writing a client for the [Firebase REST API](https://www.firebase.com/docs/rest/api/) that returns JSON Object but also JSON literals when you query leaf nodes.

> I'm worried that you could be using the library incorrectly...

You're right to be worried :) I found some [very ugly workaround](https://github.com/proppy/firebase-arduino/blob/arduino/src/FirebaseObject.cpp#L27) that I should probably not show to you :)

Merci pour la response rapide!

bblanchon, 2016-04-27:
Thanks for this information.

I think, I'll implement this one myself, and bundle it in 5.3 with other improvements I'm working on.
Hopefully it will be done by the end of the week.

proppy, 2016-04-27:
@bblanchon feel free to add me one the PR, I'm curious how complicated it would be.

(From a quick glance at the code, it seems surfacing `JsonParser::parseAnythingTo` on `JsonBuffer` would do the trick, but it's very likely I'm missing many details).

bblanchon, 2016-04-27:
That's it.
Plus a bunch of unit tests :wink:

closed 2016-04-30 by bblanchon

proppy, 2016-05-02:
Thanks a lot!