bblanchon/ArduinoJson #268: JsonObject.printTo(chart[ ], len) results in missing "}" when printed to Serial

arashmarzi, 2016-04-28:
I notices a slight bug, probably a one line fix somewhere. When the `.printTo(...)` method is used get a char array of the json object and then print it to serial, it results in a line that has a missing "}" at the end. I use the sample code in addition to live sensor data collection but it shouldnt make a difference. Note: prettyPrintTo(..) does a good job but i do not need the extra new line characters and indents. I have attached the code: 
```

// Copyright Benoit Blanchon 2014-2016
// MIT License
//
// Arduino JSON library
// https://github.com/bblanchon/ArduinoJson
// If you like this project, please add a star!

#include <ArduinoJson.h>
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_BMP085_U.h>

Adafruit_BMP085_Unified bmp = Adafruit_BMP085_Unified(10085);

void displaySensorDetails(void)
{
  sensor_t sensor;
  bmp.getSensor(&sensor);
  Serial.println("------------------------------------");
  Serial.print  ("Sensor:       "); Serial.println(sensor.name);
  Serial.print  ("Driver Ver:   "); Serial.println(sensor.version);
  Serial.print  ("Unique ID:    "); Serial.println(sensor.sensor_id);
  Serial.print  ("Max Value:    "); Serial.print(sensor.max_value); Serial.println(" hPa");
  Serial.print  ("Min Value:    "); Serial.print(sensor.min_value); Serial.println(" hPa");
  Serial.print  ("Resolution:   "); Serial.print(sensor.resolution); Serial.println(" hPa");
  Serial.println("------------------------------------");
  Serial.println("");
  delay(500);
}

void setup() {
  Serial.begin(9600);
  while (!Serial) {
    // wait serial port initialization
  }

  if (!bmp.begin())
  {
    /* There was a problem detecting the BMP085 ... check your connections */
    Serial.print("Ooops, no BMP085 detected ... Check your wiring or I2C ADDR!");
    while (1);
  }

  /* Display some basic information on this sensor */
  displaySensorDetails();

  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();

  float temperature;
  bmp.getTemperature(&temperature);

  float seaLevelPressure = 1015.6; // Average Ottawa SENSORS_PRESSURE_SEALEVELHPA;

  float bTemperature;
  float bPressure;
  float bAltitude;

  /* Get a new sensor event */
  sensors_event_t event;
  bmp.getEvent(&event);

  if (event.pressure)
  {
    bmp.getTemperature(&temperature);

    seaLevelPressure = 1015.6; // Average Ottawa SENSORS_PRESSURE_SEALEVELHPA;

    bTemperature = temperature;
    bPressure = event.pressure;
    bAltitude = bmp.pressureToAltitude(seaLevelPressure, event.pressure);
  
  // Add values in the object
    //
    // Most of the time, you can rely on the implicit casts.
    // In other case, you can do root.set<long>("time", 1351824120);
    root["sensor"] = "bmp";
    root["time"] = 1351824120;

    JsonObject& data = root.createNestedObject("data");
    data["T"] = bTemperature;
    data["P"] = bPressure;
    data["A"] = bAltitude;

    int len = root.measureLength();
    char rootstr[len];
    root.printTo(rootstr, len);
    
    //root.printTo(Serial);
    // This prints:
    // {"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}

    Serial.println(rootstr);

    root.prettyPrintTo(Serial);
    
  } else {}

    // This prints:
    // {
    //   "sensor": "gps",
    //   "time": 1351824120,
    //   "data": [
    //     48.756080,
    //     2.302038
    //   ]
    // }
  }
  void loop() {
    // not used in this example
  }
```

[image]

arashmarzi, 2016-04-28:
I was able to fix it by allocating an extra element of space to the char array, essentially you need an array that is JsonObject.measuredLength()+1

bblanchon, 2016-04-28:
Hi Arash,

`size_t JsonObject::measureLength()` returns the *length of the string*, i.e. the number of characters composing the string. It doesn't count the zero-terminator.

`JsonObject::printTo(char* buf, size_t size)` takes the *size of the buffer*. It always adds a zero-terminator at the end of the string to prevent a buffer overrun.

That is why you need to allocate a buffer which `size = len+1`.

BTW, the documentation was missing so I added it: [API Reference: JsonObject::measureLength()](https://github.com/bblanchon/ArduinoJson/wiki/API-Reference#jsonobjectmeasurelength).

arashmarzi, 2016-04-29:
Thanks for updating the documentation and your response. Great work on the library!

closed 2016-04-29 by arashmarzi

bblanchon, 2016-04-29:
You're welcome Arash, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: