bblanchon/ArduinoJson #271: Positioning key-value pair at specific index

arashmarzi, 2016-05-02:
I'm in an interesting predicament

In my scenario, I need to create a data packet that gets created from an Arduino, transmitted (via tx/rx) to an ESP8266, which then transmitted to a Mosquitto queue on a Raspberry Pi, which is then picked up by a python subscriber.

Im using a 100 byte JsonBuffer to create a data packet that contains a nested object and a corresponding CRC32 value like this {"data":{"T":25.00,"P":1010.59,"A":41.70},"C":"11fbd26e"}, where T, P, and A are sensor values and C is the CRC32 value. The ESP8266 is told to look out for 100 bytes of data coming from the tx/rx.

Even though my arduino transmits a single packet at a time to the ESP8266, the python subscriber ends with several data packets, almost in a backlogged manner with the last packet being a partial packet, like this:  

{"data":{"T":26.00,"P":1002.59,"A":41.70},"C":"11fbfd2e"} {"data":{"T":25.00,"P":1010.59,"A":41.70},"C":"11fbd26e"} {"data":{"T":26.00,"P":109.59,"A":4

I think what is happening is that all 100 bytes of the ESP8266's serial buffer are not being used up or something so it waits until its full before it transmits data to the Mosquito queue, causing partial packets. I do not know if I will know the fixed data size of my packet, or if there is some way to guarantee a fixed/predictable packet size. I know about the JSON_OBJECT_SIZE approach to setting the JsonBuffer size.

One approach to solving my problem would be to have the length of the data + CRC32 value together stored as a variable at the head of the JSON packet like this {"L":XX, "data":{"T":26.00,"P":1002.59,"A":41.70},"C":"11fbfd2e"} so that I can manually parse out and verify from data onward on the Python subscriber side. Otherwise I would have to use an IndexOf("{")/IndexOf("}") approach, but there are too many comparisons of { and } positions to make this approach practical, especially with the nested object and unpredictable cut off point in transmission.

In order to use the length approach I need to be able to put L(ength) at the head of the root object, but consequently I need to create and add the nested object and CRC32 value to the root object before I can add the length, which then ends up at the end of the root object when added. 

Is there are way to re-position the index of a key-value pair, to do  {"L":XX, "data":{"T":26.00,"P":1002.59,"A":41.70},"C":"11fbfd2e"} instead of  {"data":{"T":26.00,"P":1002.59,"A":41.70},"C":"11fbfd2e", "L":XX}? Or do I have to create temp JsonBuffer and add all the data points, calculate the length then add the length first to a new JsonBuffer object, then add the same data, then transmit the length-included packet? Or is there a better way to approach this scenario?

bblanchon, 2016-05-02:
Hi Arash,

It looks to me that the ESP8266 is implementing the [Nagle algorithm](https://en.wikipedia.org/wiki/Nagle%27s_algorithm).

I think you should try to call [WifiClient::setNoDelay(true)](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/WiFiClient.h#L73) before looking for complex workarounds.

arashmarzi, 2016-05-03:
If you deem it applicable, feel free to set this issue as another question

I looked into the Nagle Algorithm, Im assuming that you are implying the ESP8266 uses this algorithm in order to communicate with Mosquitto server as it is using TCP for sending packets. Nagles would entail that the ESP8266 waits until there is a certain amount of data then sends it off to the Mosquitto queue; is this what you meant?

I have not worked with the esp8266/Arduino library before. Where does it fit in to the architecture? Presently, Im running regular Arduino/cpp/h files on my Arduino Nano and NodeMCU on the ESP8266. Am I supposed to use the esp8266/Arduino library as a firmware or do i run it on my arduino to interface with the esp8266 over serial? My concern would be that a library like that would be too much overhead for the Nano

bblanchon, 2016-05-03:
The Nagle algorithm is implemented in the LWIP library which is used the ESP8266 core.

See [ESP8266WiFi/src/include/ClientContext.h, line 108](https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266WiFi/src/include/ClientContext.h#L108):

```c++
void setNoDelay(bool nodelay){
    if(!_pcb) return;
    if(nodelay) tcp_nagle_disable(_pcb);
    else tcp_nagle_enable(_pcb);
}
```

I'm trying to help you as much as I can, but you have to realize these are not issues related to ArduinoJson.
To get the best answer, you should ask authors of the ESP8266 core or ask a question on StackOverflow.

closed 2016-05-11 by bblanchon