bblanchon/ArduinoJson #272: Create Array from root

hallard, 2016-05-02:
I'm using ArduinoJSon with new AsyncTCPWebServer for ESP8266 and it's working very well, thanks for the great job.

Anyway, `<AsyncJson.h>` is a great handler for AduinoJson object but I'm facing one problem that I can't solve, but may be I'm doing something wrong.
I would like to start my JSON by an array, of course I can create root then nested array doing something like that without any problem (2 nested arrays)
```json
{ 
	"files":[
	{"na":"/css/nrjmeter.css.gz","va":"24325"}
	,{"na":"/favicon.ico","va":"1150"}
	,{"na":"/js/main.js.gz","va":"6131"}
	,{"na":"/version.json","va":"57"}
	],
	"spiffs":[
	{"Total":957314, "Used":163652, "ram":21440}
	]
}
```

But sometimes I need to have direct array as follow (for bootstraptable) and `createArray` seems not to be supported on root object (error at compile time) only `createNestedArray` works
```json
[
  {"ssid":"FreeWifi_secure","rssi":-59,"enc":"????","chan":1},
  {"ssid":"HOME-FREEBOX","rssi":-60,"enc":"WPA2","chan":1},
]
```

Of course I could `CreateNestedArray("")` on root object but in this case I've got something like this
```json
{"nullptr": [
  {"ssid":"FreeWifi_secure","rssi":-59,"enc":"????","chan":1},
  {"ssid":"HOME-FREEBOX","rssi":-60,"enc":"WPA2","chan":1},
  ]
}
```
The solution I was using was to do a `createArray()` on JSONbuffer and it worked, but since I'm using Async wrapper now, it create a root JSon object, not an array.

Any tips would be appreciated

bblanchon, 2016-05-02:
Bonjour Charles,

I don't know what `<AsyncJson.h>` is.
But concerning, ArduinoJson, you just need to call `JsonBuffer::createArray()`:

```c++
DynamicJsonBuffer jsonBuffer;
JsonArray& root = jsonBuffer.createArray();
JsonObject & signal1 = root.createNestedObject();
signal1["ssid"] = "FreeWifi_secure";
// ...
```

hallard, 2016-05-02:
Benoit, 
thanks for your quick reply, sorry about that AsyncJson is part of the great project ESPAsyncWebServer https://github.com/me-no-dev/ESPAsyncWebServer done by me-no-dev
source file is [here](https://github.com/me-no-dev/ESPAsyncWebServer/blob/master/src/AsyncJson.h)

Yeah the tips you give me works fine but since I migated my code with the Whole project ESPAsyncWebServer I don't have "hand" on the object creation, it's created in the class and I only got pointer on JSONObject and not a JSONArray (here the source declaration from AsyncJson)
```arduino
  public:
    AsyncJsonResponse(): _isValid{false} {
      _code = 200;
      _contentType = "text/json";
      _root = _jsonBuffer.createObject();
    }
```
So in this case, there is no way to create simple (so not nested) array from the root object ?

bblanchon, 2016-05-02:
Charles,
I think you should make a PR to @me-no-dev.
Adding another constructor taking a `JsonVariant` seems reasonable.

hallard, 2016-05-02:
Benoit,
Makes sense, I first tried there because before I was wondering this question before migration to to Async :-)

Merci, et bonne soirée

me-no-dev, 2016-05-02:
Hey guys sorry to barge in your convo :) but what is the best option here? And what are the possible types that the root can be? Can a Variant become array/object or casted as?
I have not really used the lib with async and most work in the AsyncServer is based on other user's cases and code.
If the option is only object/array then an argument can be added to the constructor to define it as array instead.

hallard, 2016-05-02:
@me-no-dev,
Glad to see you ;-)
As far as I used Benoit lib, the only need I encounter was creating an direct array (and I'm using lot of case) so from my point of view and usage, yes option for only object or array will be enough ;-)

@me-no-dev, I migrated my full project to ESPasync, works fine, got some questions but I will open case on you github later for them ;-)

me-no-dev, 2016-05-02:
@hallard in the latest commit :) maybe close the issue if that works for you. Will expect your questions :)

hallard, 2016-05-02:
Thanks,
Currently testing, seems to works with small JSON, I've got strange things with larger JSON (>750 bytes), need to check this point.
May be set declaration to false if called with no parameters (so JSONObject) to avoid breaking old users calling with no parameters like this ;-)
`AsyncJsonResponse()`

closed 2016-05-03 by bblanchon