bblanchon/ArduinoJson #273: Drop arduino String class at the FirebaseArduino layer.

ed7coyne, 2016-05-02:
We will use std::string on all deeper layers, this makes testing and code sharing much easier

bblanchon, 2016-05-03:
I'm sorry, I don't understand.
Can you be more specific?

FYI, when used outside of Arduino, `String` is just a `typedef` of `std::string`.
See [ArduinoJson/Arduino/String.hpp, line 13](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Arduino/String.hpp#L13)

ed7coyne, 2016-05-03:
Sorry, I am still new to github. I created the issue against the wrong project. This explains why I couldn't assign myself to it though :)

ed7coyne, 2016-05-03:
Although since you mentioned the typdef, would you be opposed to adding a indef guard around it, like this: https://github.com/googlesamples/firebase-arduino/blob/master/src/third-party/arduino-json-5.2/include/ArduinoJson/Arduino/String.hpp#L12

It is a somewhat common requirement to override arduino String for testing off of arduino and without this guard it makes it harder to include your library. 

I think we (firebase-arduino) won't need it anymore because we are going to stop using String in any tested classes so we will drop our define but it might be nice to provide for other library users.

closed 2016-05-03 by ed7coyne

bblanchon, 2016-05-03:
I think it would be better not to declare anything in the global namespace.

I can replace this by an `ArduinoJson::String` that would either be a typedef of `std::string` or Arduino's `String`.

What do you think?

ed7coyne, 2016-05-03:
That sounds like a pretty clean solution. Might even steal that idea and use it on our side :)