bblanchon/ArduinoJson #274: type problem on char buffer on ESP8266 ?

hallard, 2016-05-02:
Benoit,

I'm facing a strange issue on ESP8266 here is my code 

```arduino
  ... blah blah
  char buffer[32];
  JsonObject& item = arr.createNestedObject();
  sprintf_P( buffer, PSTR("%ld ms"), wifi_connect_time);  
  DebugF( "Wifi established in "); Debugln(buffer);
  item["na"] = "Wifi established in";
  //item["va"] = (char*)buffer; same thing with this one
  item["va"] = buffer; 
  String testStr;
  arr.printTo(testStr); 
  Debugf( "JSON=%s",testStr.c_str());
  ...blha blah
```
debug ouput is (using printf in Debug Macro)
```Wifi established in 3726 ms```
so my char buffer is correct this is my var `wifi_connect_time` value but the json output is
```json
JSON={"na":"Wifi established in","va":"17%"}
```
Any idea ?

bblanchon, 2016-05-03:
No idea, sorry.
Are you sure this program is enough to reproduce the bug ?
I suppose that the source of the problem is not shown here.
In particular, I suspect that you are reusing `buffer` or having it out of scope before calling `printTo()`.

Please have a look at:

* [Avoiding pitfalls: Do not assume that strings are copied](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#6-do-not-assume-that-strings-are-copied)

hallard, 2016-05-03:
Benoît,
Got it, Buffer is out of scope for sure, sorry

closed 2016-05-03 by hallard