bblanchon/ArduinoJson #277: Putting the encoded string into a variable

thewhiterabbit, 2016-05-10:
Is it possible to put the json encoded string into a char* variable?

If so, how?

I wish for something like this:

char* jsonStr = object.jsonString;

mweisshaupt1988, 2016-05-11:
Yes, I use this code to write it to a char*: 
```
size_t bufferSize = object.measurePrettyLength() * sizeof(char) + 1;
charBuffer = (char*) malloc(bufferSize);
object.prettyPrintTo(charBuffer, bufferSize);
```

You can also use functions that don't prettyfy the String so it is harder to read for humans but saves some memory.

bblanchon, 2016-05-11:
Even I the solution from @mweisshaupt1988 works, the preferred ways are:

```c++
char json[128];
object.printTo(json, sizeof(json));
```

or 

```c++
String json;
object.printTo(json);
```

Use the first one if you prefer static memory allocation, or the second if you accept dynamic memory allocation.

@thewhiterabbit, I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

thewhiterabbit, 2016-05-11:
Now I understand. The printTo function can assign the value to the variable. I am used to assigning values differently. Thank you both for the code examples. @bblanchon I simply missed that you were assigning value differently than I am used to, using printTo. I saw this in the documentation. It is a good documentation. Sorry for my misunderstanding. I am not used to the c/c++ languages yet.

bblanchon, 2016-05-11:
I could have implemented something like that:

```c++
String json = object;
```

but this poses 4 problems:

1. How to distinguish between `printTo()` and `prettyPrintTo()` ?
2. It's an implicit conversion (C++03 didn't support explicit user-defined conversion)
3. I don't believe that a good practice since it relies upon dynamic memory allocation
4. People would be tempted to write `Serial.println(object)`, thereby ruining the performance.

Then, I could have solved problems 1 and 2 with something like:

```c++
String json = object.toString();
String prettyJson = object.toPrettyString();
```

But problem 3 &  4 remains.

Plus, it's not a secret that [I hated the `String` class](http://blog.benoitblanchon.fr/arduino-json-v5-0) :wink:.

closed 2016-05-13 by bblanchon