bblanchon/ArduinoJson #278: mapping json into Tree-Object with seperate nodes

Pfannex, 2016-05-13:
Hi,

is there a way to map a json into something like a Tree-Object with seperate nodes?

Greatings
Pf@nne

bblanchon, 2016-05-13:
Hi,
I'm sorry, I don't understand the question?
Are you talking about compile time introspection, like boost::fusion?
Regards,
Benoit

Pfannex, 2016-05-13:
Hi Benoit,

OK, I try to describe what I want to do.

For my MQTT communication I want to read the topicTree from a json-file.
The Topics-Strings inside the json are unknown. 

I want to map the topicTree into a TreeNode-Object with some more functions then your json-library like:

- has child
- parent
- count nodes in each level

I want to get access to the whole tree objects by index not only arrays.

E.g.

root[0][0].key
root[0][0].value
root[0][0].hasChild
root[0][0].countChilds
root[0][0].getParent

Greatings
Pf@nne

bblanchon, 2016-05-14:
It looks like you just want to enumerate all the key-value pairs of an object.

This can be done by using iterators:

```c++
JsonObject& object = jsonBuffer.parseObject(json);

for(JsonObject::iterator it=object.begin(); it!=object.end(); ++it) 
{
    const char* key = it->key;
    JsonVariant value = it->value;

    if (value.is<JsonObject>()) {
        JsonObject& nestedObject = value;
        // etc.
    }
}
```

bblanchon, 2016-05-18:
@Pfannex Did this work for you?

Pfannex, 2016-05-18:
@bblanchon 

I changed my json MQTT-Topic from object to array.
So direct access by index is given.....

```
root[0][1]
```

Your hint will also work probably, thanks for this!

Pfannex, 2016-05-22:
Hi,

access by index works fine, but sub indexes are shifted +1.

eg:

json stringArray

```
[
    "E0",
    "E1",
    "E2",
    "E3",
    	[
    	"E3.0",
    	"E3.1",
    		[
    		"E3.1.0",
    		"E3.1.1",
    		]
    	]
]
```

to get access it is required to use sub indexes +1.

```
		Serial.println(root[0].asString());
		Serial.println(root[1].asString());
		Serial.println(root[2].asString());
		Serial.println(root[3].asString());
		Serial.print("   "); Serial.println(root[4][0].asString());
		Serial.print("   "); Serial.println(root[4][1].asString());
		Serial.print("      "); Serial.println(root[4][2][0].asString());
		Serial.print("      "); Serial.println(root[4][2][1].asString());
```

Of course this is because root[0] to root[3] are "real" Elements while Element root[4] ist an jsonArrayObject and root[4][0] ist an real Element of the jsonArrayObject root[4].

But for using the indexes to itemize the elements it is a little confusing.

Is there a way to work arround?

Greetings
Pf@ne

bblanchon, 2016-05-22:
I'm sorry, but once again, I don't understand your question.
Can you please rephrase?

Pfannex, 2016-05-22:
I want to get access to a jsonArray by calling the elemnt by index.
Basicalli this works fine, but sub arrays must be called by index +1.

See listing above.

bblanchon, 2016-05-22:
What do you mean by "sub arrays must be called by index +1"?
It looks OK to me in the code snippet.

Pfannex, 2016-05-23:
First, sorry for my bad English and your patience!

To get access to Element "E3.0" you must call array "root [3+1][0]".
Thats what I mean with "shifted".

The logical order in my mind is that a "child"Array should be called by leading index of the parent index.

At the end I will create a routine that results the full path of a Element.

eg:

```
path (root [3+1][1+1][0]) should return "E3/E3.0/E3.1.0"
              ^    ^
               Index +1
```

I hope you understand my problem....

bblanchon, 2016-05-23:
I wrote a [test reproducing this issue](https://github.com/bblanchon/ArduinoJson/blob/73897373900924c17251175d827837fa4f1020cf/test/Issue278.cpp), but everything is working as expected.

I think that you are making a mistake.
I believe that the source of your confusion is the incorrect naming of values in the array.
To me, the JSON should be:

```json
[
  "E0",
  "E1",
  "E2",
  "E3",
  [
    "E4.0",
    "E4.1",
    [
      "E4.2.0",
      "E4.2.1"
    ]
  ]
]
```

Pfannex, 2016-05-23:
> I believe that the source of your confusion is the incorrect naming of values in the array.

The naming will be correct for a json array,
but it might be not really corect for a standard "subNaming"

eg:

Textdocument with sub entries

0. first mainpoint
1. second mainpoint
  1.0 first sub of second mainpoint
  1.1 second sub of second mainpoint
    1.1.0
    1.1.1

this should be a logical and consistent naming.

json like:

root[0]
root[1]
  root[2][0]
  root[2][1]
    root[2][2][0]
    root[2][2][1]

maybe I'm the only one who's getting confused

Pfannex, 2016-06-05:
Hi, it's me again... :-)

Do you have a idea how to get the full path (connected array item names) of a specific array?

e.g.

**json**
```
[
  "E0",
  "E1",
  "E2",
  "E3",
  [
    "E4.0",
    "E4.1",
    [
      "E4.2.0",
      "E4.2.1"
    ]
  ]
]
```

`path (root [4][2][0]) `
should return "E3/E4.0/E4.2.0"

This works Step by Step:
```
	path = root[3].asString();

    	JsonArray& nestedArray0 = root[3+1].asArray();
    	path += "/";
    	path += nestedArray0[1].asString();

    	JsonArray& nestedArray1 = nestedArray0[1+1].asArray();
    	path += "/";
    	path += nestedArray1[0].asString();
```

Now I want to put it into a loop, to do this for every, also unknown, array structures.
My problem is how to overwrite a existing arrayObject?

e.g.
```
JsonArray& root = jsonBuffer.parseArray(jsonString);

for ( int i = 0 ; i < countDepth ; i++ ) {  

  JsonArray& tempArray = root[pathIndex[i]+1].asArray();

	    	Serial.println("temp--------------------------------------");
	    	tempArray.prettyPrintTo(Serial);						//works!!
	    	Serial.println("------------------------------------------");

	    	JsonArray& root = tempArray;							//didn't work
	    }
```

do you have a hint for me?
thanks and greatings from Hamburg (sunny 27°C, very hot)
Pf@nne

bblanchon, 2016-06-06:
Hi,

You can loop on all items of an array thanks to [iterators](https://github.com/bblanchon/ArduinoJson/wiki/Examples#example-3-iterators).
You can modify the arrays and objects in place.

Regards,
Benoit

closed 2016-06-16 by bblanchon