bblanchon/ArduinoJson #279: error should be surfaced when parsing variant

proppy, 2016-05-14:
Version 5.2 introduced a method to parse variant, but the [current implement](https://github.com/bblanchon/ArduinoJson/blob/36d1bb2f52be71d1ff755f4a4d42d9ed838bf67a/include/ArduinoJson/Internals/JsonParser.hpp#L32) doesn't surface the return value of  `parseAnythingTo`.

Maybe it could check for the return value and return `JsonVariant::invalid()` ?

bblanchon, 2016-05-14:
True.
I'll fix that ASAP.

bblanchon, 2016-05-15:
Hmmm....

`JsonVariant::invalid()` is not what you think.
`JsonVariant::invalid<T>()` is the value returned if you call `JsonVariant::as<T>()` but the variant cannot be casted to this type.
I have to rename this.

I think that you want is `bool JsonVariant::success()` similar to what we have for `JsonArray` and `JsonObject`:

```c++
JsonVariant variant = jsonBuffer.parse();
if (!variant.success()) {
   // error
}
```

proppy, 2016-05-15:
@bblanchon yes, would a method that just return `_type != JSON_UNDEFINED` be enough? or is the a better way to bubble up the result from `parseAnythingTo` to the variant instance.

bblanchon, 2016-05-16:
I'll just return `_type != JSON_UNDEFINED`.
I'll also rename or move `JsonVariant::invalid()`.

closed 2016-05-18 by bblanchon