bblanchon/ArduinoJson #280: Following the tutorial, but can't decode server json response?

thewhiterabbit, 2016-05-15:
I am having difficulty in getting the value '300000000' from the json response into my 'sleepTime' variable. I noticed that trying to ENCODE the doubles or floating point values sent to the server I needed to wrap them in quotes, converting them into a string, then once they get to the server I have to convert them back to floats or doubles to save them in the database.

Is the same true for all numeric based values in the DECODE process? I was able to ENCODE the longs and integers without wrapping them in quotes. Maye my assignment method is wrong, but doesn't look much different than in the tutorial. I tried to explicitly cast the response value for sleep_time into a long. Still not working.

I can tell that the value of sleep_interval is not carrying over to 'sleepTime' because deep sleep only works if I replace 'sleepTime' with '300000000' statically.

Here is the response...
{"status":1,"message":"Data successfully saved.","sleep_interval":300000000}

Here's the code...

//Libraries
#include <DHT.h>;
#include <ESP8266WiFi.h>;
#include <ArduinoJson.h>;

//Constants
#define DHTPIN 2     // what pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302)
DHT dht(DHTPIN, DHTTYPE); //// Initialize DHT sensor for normal 16mhz Arduino

// ACCOUNT SETUP CONSTANTS
const int nodeID      = ESP.getChipId();
const char* ssid      = "xxxxxx";  // WiFi Router SSID *Usually on side or bottom of router*
const char* password  = "xxxxxx";     // WiFi Router Password *Usually on side or bottom of router*
const char* apiHost   = "192.168.2.3"; // Local API server *the PCduino's assigned Hostname or IP*
  const int apiPort   = 80;

void setup()
{
  Serial.begin(115200);
  delay(100);
  // We start by connecting to a WiFi network...
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
  }

  dht.begin();

  delay(2000);
 
  Serial.print("connecting to ");
  Serial.println(apiHost);
  
  // Variables
  String url;
  String request;
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  if (!client.connect(apiHost, apiPort)) {
    Serial.println(String(apiHost) + " : connection failed");
    return;
  }

  Serial.print("Requesting URL: ");
  Serial.println(url);
  
  // Read Temp & Humdity data and store it to variables...
  String humidity = String(dht.readHumidity());
  String airTemp = String(dht.readTemperature());
  
  StaticJsonBuffer<200> jsonBuffer;

  JsonObject& device  = jsonBuffer.createObject();
  device["node_id"]   = nodeID;
  device["sensors"]   = "DHT22";

  JsonObject& data = device.createNestedObject("data");
  data["humidity"]  = humidity;  // 6 is the number of decimals to print
  data["air_temp"]  = airTemp;   // if not specified, 2 digits are printed

  device.printTo(request);
  // We now create a URI for the request
  url = "/?request=" + urlencode(request);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + apiHost + "\r\n" + "Connection: close\r\n\r\n");
  delay(100);
  
  // Read all the lines of the reply from server and print them to Serial
  String response;
  while(client.available()){
    response = client.readString();
    ///// response returns -> {"status":1,"message":"Data successfully saved.","sleep_interval":300000000} ////
  }
  JsonObject& responseObj = jsonBuffer.parseObject(response);
  long sleepTime          = long(responseObj["sleep_interval"]); ///// THE PART THAT IS NOT WORKING /////
  
  ESP.deepSleep(sleepTime);
  delay(1000);
}
//
//
void loop()
{

}
//
//
String urldecode(String str)
{
    
    String encodedString="";
    char c;
    char code0;
    char code1;
    for (int i =0; i < str.length(); i++){
        c=str.charAt(i);
      if (c == '+'){
        encodedString+=' ';  
      }else if (c == '%') {
        i++;
        code0=str.charAt(i);
        i++;
        code1=str.charAt(i);
        c = (h2int(code0) << 4) | h2int(code1);
        encodedString+=c;
      } else{
        
        encodedString+=c;  
      }
      
      yield();
    }
   return encodedString;
}

String urlencode(String str)
{
    String encodedString="";
    char c;
    char code0;
    char code1;
    char code2;
    for (int i =0; i < str.length(); i++){
      c=str.charAt(i);
      if (c == ' '){
        encodedString+= '+';
      } else if (isalnum(c)){
        encodedString+=c;
      } else{
        code1=(c & 0xf)+'0';
        if ((c & 0xf) >9){
            code1=(c & 0xf) - 10 + 'A';
        }
        c=(c>>4)&0xf;
        code0=c+'0';
        if (c > 9){
            code0=c - 10 + 'A';
        }
        code2='\0';
        encodedString+='%';
        encodedString+=code0;
        encodedString+=code1;
        //encodedString+=code2;
      }
      yield();
    }
    return encodedString;
    
}

unsigned char h2int(char c)
{
    if (c >= '0' && c <='9'){
        return((unsigned char)c - '0');
    }
    if (c >= 'a' && c <='f'){
        return((unsigned char)c - 'a' + 10);
    }
    if (c >= 'A' && c <='F'){
        return((unsigned char)c - 'A' + 10);
    }
    return(0);
}

bblanchon, 2016-05-16:
I'd like to help you, but there is too much information in your question.
Can you please isolate the relevant parts?

bblanchon, 2016-05-18:
@thewhiterabbit I've just noticed that your reusing a tiny `StaticJsonBuffer`, you should try to increase the size, or simply switch to a `DynamicJsonBuffer`.

Please have a look at:

* [Avoiding Pitfalls: Make StaticJsonBuffer big enough](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#1-make-staticjsonbuffer-big-enough)
* [Avoiding Pitfalls: Don't reuse the same JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-jsonbuffer)
* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)
* [FAQ: What's the best way to use the library?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)
* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

thewhiterabbit, 2016-05-19:
I have to update OS X so I can try to get serial debugging to work again before testing your suggestions. Thank you for the reply, and suggestions.

thewhiterabbit, 2016-05-20:
I've successfully updated OS X to El Capitan and can now get debugging from serial.

After reading your documents...

My updated code:

void setup()
{
  Serial.begin(57600);
  dht.begin();
  delay(2000);

  
  // We start by connecting to a WiFi network...
  WiFi.begin(ssid, password);
  
  while (WiFi.status() != WL_CONNECTED) {
    Serial.println(".");
    delay(500);
  }
   
  Serial.print("connecting to ");
  Serial.println(apiHost);
  
  // Variables
  String url;
  String request;
  
  // Use WiFiClient class to create TCP connections
  WiFiClient client;
  if (!client.connect(apiHost, apiPort)) {
    Serial.println(String(apiHost) + " : connection failed");
    return;
  }

  //Serial.print("Requesting URL: ");
  //Serial.println(url);

  // Read Temp & Humdity data and store it to variables...
  float humidity = dht.readHumidity();
  float airTempC = dht.readTemperature();
  float airTempF = dht.readTemperature(true);
  
  DynamicJsonBuffer jsonBuffer;

  JsonObject& device  = jsonBuffer.createObject();
  device["node_id"]   = nodeID;
  device["sensors"]   = "DHT22";

  JsonObject& data = device.createNestedObject("data");
  data["humidity"]  = String(humidity);
  data["air_tempC"]  = String(airTempC);
  data["air_tempF"]  = String(airTempF);

  device.printTo(request);
  // We now create a URI for the request
  url = "/?request=" + urlencode(request);
  
  // This will send the request to the server
  client.print(String("GET ") + url + " HTTP/1.1\r\n" + "Host: " + apiHost + "\r\n" + "Connection: close\r\n\r\n");
  delay(1000);
  
  // Read all the lines of the reply from server and print them to Serial
  String response;
  long sleepTime = 0;
  while(client.available()){
    delay(1000);
    response = client.readString();
  }
  Serial.println("Response: "+response);
  DynamicJsonBuffer jsonBuffer2;
  JsonObject& responseObj = jsonBuffer2.parseObject(response);
  if (!responseObj.success()){
    Serial.println("B0rken!");
  }else{
    long hoohoo = responseObj["sleep_interval"];
    Serial.println(hoohoo);
  }
  sleepTime          = responseObj["sleep_interval"];
  if (sleepTime == 0) {
    sleepTime = 5000000;
  }
  ESP.deepSleep(sleepTime);
  delay(1000);
}
//
//
void loop()
{

}
//
//
String urldecode(String str)
{
    
    String encodedString="";
    char c;
    char code0;
    char code1;
    for (int i =0; i < str.length(); i++){
        c=str.charAt(i);
      if (c == '+'){
        encodedString+=' ';  
      }else if (c == '%') {
        i++;
        code0=str.charAt(i);
        i++;
        code1=str.charAt(i);
        c = (h2int(code0) << 4) | h2int(code1);
        encodedString+=c;
      } else{
        
        encodedString+=c;  
      }
      
      yield();
    }
   return encodedString;
}

String urlencode(String str)
{
    String encodedString="";
    char c;
    char code0;
    char code1;
    char code2;
    for (int i =0; i < str.length(); i++){
      c=str.charAt(i);
      if (c == ' '){
        encodedString+= '+';
      } else if (isalnum(c)){
        encodedString+=c;
      } else{
        code1=(c & 0xf)+'0';
        if ((c & 0xf) >9){
            code1=(c & 0xf) - 10 + 'A';
        }
        c=(c>>4)&0xf;
        code0=c+'0';
        if (c > 9){
            code0=c - 10 + 'A';
        }
        code2='\0';
        encodedString+='%';
        encodedString+=code0;
        encodedString+=code1;
        //encodedString+=code2;
      }
      yield();
    }
    return encodedString;
    
}

unsigned char h2int(char c)
{
    if (c >= '0' && c <='9'){
        return((unsigned char)c - '0');
    }
    if (c >= 'a' && c <='f'){
        return((unsigned char)c - 'a' + 10);
    }
    if (c >= 'A' && c <='F'){
        return((unsigned char)c - 'A' + 10);
    }
    return(0);
}

The serial output:

connecting to 192.168.2.3
Response: HTTP/1.1 200 OK
Date: Fri, 20 May 2016 14:53:43 GMT
Server: Apache/2.4.18 (Unix) PHP/5.5.34 LibreSSL/2.2.6
X-Powered-By: PHP/5.5.34
Content-Length: 77
Connection: close
Content-Type: text/html

{"status":1,"message":"Data successfully saved.","sleep_interval":300000000}
B0rken!

As you can see the responseObj.success() is failing...

So I am thinking maybe I am not using correct success check or it truly is failing the success check because the response is not able to be parsed because the JSON returned is preceded by the HTTP header information.

How can I strip the HTTP header away to only parse the JSON response?

Please forgive me as I am still learning C++.

bblanchon, 2016-05-20:
You should have a look at the [JsonHttpClient](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino) example.

In particular, have a look at the [skipResponseHeaders()](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino#L100) function:

```c++
// Skip HTTP headers so that we are at the beginning of the response's body
bool skipResponseHeaders() {
  // HTTP headers end with an empty line
  char endOfHeaders[] = "\r\n\r\n";

  client.setTimeout(HTTP_TIMEOUT);
  bool ok = client.find(endOfHeaders);

  if (!ok) {
    Serial.println("No response or invalid response!");
  }

  return ok;
}
```

thewhiterabbit, 2016-05-20:
Thank you very much for your help @bblanchon, I could not have gotten this far without it.

I see that this is using the Ethernet library. I am using the ESP8266WiFi library.

How can I apply those functions you linked, to the WiFi library?

I am imagining to change the functions 'skipResponseHeaders()' and/or 'readReponseContent()' , but I am not understanding how 'response' could be assigned/reassigned. Currently I am getting the entire response including headers in 'response', so can I simply paste those functions you linked to into my code and call readReponseContent(response) or do I need to do 'response = readReponseContent(response)' or am I completely misunderstanding the functions?

bblanchon, 2016-05-20:
I didn't try myself, but I believe you can reuse the `WifiClient` from the ESP8266 core as a drop-in replacement for `EthernetClient`.

Please realize that this question is not related to ArduinoJson anymore, so this is probably not the right place to ask that kind of question.

thewhiterabbit, 2016-05-20:
OK, thank you again @bblanchon I appreciate your help very much. I also appreciate this library, it is well engineered. I will try that and if I have any more trouble I will seek help in a relevant forum. :)

closed 2016-05-20 by thewhiterabbit

bblanchon, 2016-05-20:
You're welcome.

I hope your project will work.
Don't hesitate to add a link in [Wiki: Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: