bblanchon/ArduinoJson #281: 'class ArduinoJson::JsonObject' has no member named 'add'

DrCord, 2016-05-19:
JsonObject.add doesn't work even though it shows in the documentation as existing. It is needed to be able to add numeric values and set the decimal precision.

It is shown in the documentation on this page:

https://github.com/bblanchon/ArduinoJson/wiki/Encoding%20JSON

bblanchon, 2016-05-20:
I'm sorry, it's a mistake in the documentation.

`JsonObject::add()` was renamed `JsonObject::set()` in [version 5.0.0](https://github.com/bblanchon/ArduinoJson/blob/master/CHANGELOG.md#v500) to be consistent with `JsonArray`

I'll update the wiki.
Thank you for pointing this out.

closed 2016-05-20 by bblanchon

DrCord, 2016-05-21:
great, that worked like a charm, thanks for your quick response!

bblanchon, 2016-05-21:
You're welcome.
Thank you for using ArduinoJson.