bblanchon/ArduinoJson #282: Sim Shield Arduino

vratsasg, 2016-05-20:
Hello Sir I would like to ask you if i can use this library to send json to my webapp through Arduino Gsm Shield V2.For example a simple Post with few analog sensors.
1.Take sensors measures
2.Prepare json
3.Send json via GSMClient  something like that as shown below

  GSMClient client;
  char server[] = "78.46.70.93:8080"; // the base URL
  int port = 80; // the port, 80 for HTTP
  if (client.connect(server, port)){
     
    writeResponse(client, json)

 }
void writeResponse(EthernetClient& client, JsonObject& json) {
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: application/json");
  client.println("Connection: close");
  client.println();

  json.prettyPrintTo(client);
}

vratsasg, 2016-05-20:
not EthernetCliend at function writeResponse but GSMCliend& client

bblanchon, 2016-05-21:
Hi,
I didn't try it myself, but after looking at the [tutorial](https://www.arduino.cc/en/Tutorial/GSMExamplesWebClient), I think it should work.

BTW, you can make the code more generic, this way:

```c++
void writeResponse(Print& client, JsonObject& json) {
  client.println("HTTP/1.1 200 OK");
  client.println("Content-Type: application/json");
  client.println("Connection: close");
  client.println();

  json.prettyPrintTo(client);
}
```

Now it works with EthernetClient, GSMClient, and Serial.

vratsasg, 2016-05-23:
thanks bblanchon for your help and your library.i Use a tomcat server and arduino gsm shield V2.
Post a JSOn to a web server this code works for me.For a Json like this {"humidity":50,
"itemp":40,"wtemp":30,"soil":20}

`#define PINNUMBER "XXXX"   //pin number of sim card

// APN information obrained from your network provider
#define GPRS_APN       "internet.vodafone.gr" // replace with your GPRS APN   greece vodafone
#define GPRS_LOGIN     ""    // replace with your GPRS login
#define GPRS_PASSWORD  "" // replace with your GPRS password

char server[] = "78.46.70.93"; // the base URL
int port = 8080; // the port, 80 for HTTP   its a tomcat server
 StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["humidity"] = 50;
  root["itemp"] = 40;
  root["wtemp"] = 30;
  root["soil"] = 20;

 if (client.connect(server, port)) {
    Serial.println("connected");
    client.println("POST /path HTTP/1.0");
    client.println("Host: 78.46.70.93");
    client.println("User-Agent: Arduino/1.0");
    client.println("Connection: close");
    client.println("Content-Type: application/json;charset=UTF-8");
    client.print("Content-Length: ");
    client.println(root.measureLength());
    client.println();
    root.printTo(client);
}`

bblanchon, 2016-05-23:
You're welcome @vratsasg, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-05-23 by bblanchon