bblanchon/ArduinoJson #283: uint? Full 32 bits?

shovemedia, 2016-05-22:
I'll admit being a C++ noob, but I can't for the life of me figure out how to parse a full 32 bit number into a uint. For values greater than 2^31, I end up with 2147483647. I can use a double and get close (the least significant bits are off) but ...

Basically, I'm trying different variations of 
const uint a = jsonObj.get<uint32>("myInt");

Thanks for the lib -- I definitely owe you a donation already.

shovemedia, 2016-05-22:
Upgraded to the latest and seems to be working after manually adding:
using namespace std;
to fix isnan and isinf issues.

bblanchon, 2016-05-22:
The `uint` issue has been fixed in [v5.3.0](https://github.com/bblanchon/ArduinoJson/blob/master/CHANGELOG.md#v530).

But I'm quite surprised you needed to add `using namespace std`, what compiler are you using?

shovemedia, 2016-05-22:
Just using the Arduino IDE. Does the ESP8266 board matter? (Adafruit Huzzah)

bblanchon, 2016-05-22:
I opened a new issue for that new topic.

closed 2016-05-22 by bblanchon