bblanchon/ArduinoJson #284: Compilation error: isnan() and isinf() not found

bblanchon, 2016-05-22:
See issue #283.
This happens on Arduino IDE with ESP8266 core.

bblanchon, 2016-05-22:
@shovemedia I didn't manage to reproduce this bug on my Windows machine.
I downloaded the latest Arduino 1.6.9 and the latest ESP8266 core 2.2.0.

Do you have any clue?

shovemedia, 2016-05-22:
I'm on OSX, but that probably doesn't matter.
When I googled for the err message, I got a hit on WifiManager which I'm also using and is where I got the hint to add "using namespace std".

bblanchon, 2016-05-23:
@shovemedia Does this happen when you build one of the ArduinoJson examples?

bblanchon, 2016-06-04:
@shovemedia Did you have the opportunity to build one of the example alone?

domingosl, 2016-06-05:
Hello, I'm also having this issue on Windows, I'm also using the WifiManager:

.pioenvs\esp12e\ArduinoJson/include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp:29:17: error: 'isnan' was not declared in this scope

Any clues?

bblanchon, 2016-06-05:
Hi,
There is a line in the changelog of WifiManager that states:

> ##### 0.12
> * fixed incompatibility with other libs using isnan and other std:: functions without namespace

Also there is this commit: ["reverted std::sort that caused problems with isnan and other none namespeced used function calls"](https://github.com/tzapu/WiFiManager/commit/44e828e77a2454a38371f7d5fb64424acebeb676)

So I beleive it's fixed in version 0.12.
Can you check the version please?

domingosl, 2016-06-05:
@bblanchon I have a very modified (by me) version of it. But looking at the commit you point out I was able to identify the problem on the wifiManager library, I just have the c++ cryptic compiler errors...

Any way I believe  the problem from @shovemedia is the same as mine. 

Thank you @bblanchon !

bblanchon, 2016-06-06:
You're welcome.
Thank you for using ArduinoJson.

vlast3k, 2016-06-29:
I was getting the same problem with a sketch using ESP8266, but not using WiFiManager. Compilation of the examples worked (with platform ESP8266)
When i added
using namespace std;
in my main sketch file the compilation went through :)

bblanchon, 2016-06-29:
@vlast3k Can you please try to insert the `using namespace std;` inside [`Polyfills::isNaN()`](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Polyfills/isNaN.hpp#L42) and [`Polyfills::isInfinity()`](https://github.com/bblanchon/ArduinoJson/blob/master/include/ArduinoJson/Polyfills/isInfinity.hpp#L40)?

If that works, it could be a more general solution that I could integrate into the next version.
Thanks.

vlast3k, 2016-06-29:
@bblanchon thanks, it compiled successfully this way.
I changed in isInfinity.hpp
from line 39
```
template <typename T>
bool isInfinity(T x) {
  using namespace std;
  return isinf(x);
}
```

and in isNaN.hpp
from line 41
```
template <typename T>
bool isNaN(T x) {
  using namespace std;
  return isnan(x);
}
```

bblanchon, 2016-06-29:
Thank you very much. I'll add that ASAP.

bblanchon, 2016-06-30:
Fixed in v5.6.2

proppy, 2016-07-09:
I got a compilation error too (with 5.6.2) with some source that include `stdint.h` in the same translation unit:
```
../Polyfills/math.hpp:73:17: error: call of overloaded ‘isinf(double&)’ is ambiguous
```
And I had to add the following patch to make it compiles:
```
diff --git a/src/third-party/arduino-json-5.6.2/include/ArduinoJson/Polyfills/math.hpp b/src/third-party/arduino-json-5.6.2/include/ArduinoJson/Polyfills/math.hpp
index 0abc73e..de7b78e 100644
--- a/src/third-party/arduino-json-5.6.2/include/ArduinoJson/Polyfills/math.hpp
+++ b/src/third-party/arduino-json-5.6.2/include/ArduinoJson/Polyfills/math.hpp
@@ -59,7 +59,7 @@ bool isNaN(T x) {
   using namespace std;
 #endif
 
-  return isnan(x);
+  return ::isnan(x);
 }
 
 template <typename T>
@@ -70,7 +70,7 @@ bool isInfinity(T x) {
   using namespace std;
 #endif
 
-  return isinf(x);
+  return ::isinf(x);
 }
 
 #if defined(__GNUC__)
```

bblanchon, 2016-07-10:
Hi Johan,

Sorry, I missed this message because the issue was closed.

Can you please give a try to branch [issue284-3](https://github.com/bblanchon/ArduinoJson/tree/issue284-3)?
BTW, what compiler are you using?

bblanchon, 2016-07-17:
@proppy Did you have a chance to give a try to branch [issue284-3](https://github.com/bblanchon/ArduinoJson/tree/issue284-3)?

proppy, 2016-07-19:
@bblanchon sorry for the late followup. Yes, this branch works better for me.

bblanchon, 2016-07-19:
@proppy Thanks!

closed 2016-07-19 by bblanchon