bblanchon/ArduinoJson #286: parseObject() keeps failing

uzbhutta, 2016-05-25:
Hi there,

I am trying to run a sample parsing program on the Particle Photon. I have the following code:

`#include "application.h"
#include <ArduinoJson.h>

char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";

void setup() {
  Serial.begin(9600);
}

void parseJson() {
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
  }

  const char* output = root["sensor"];
  Serial.println(output);
}

void loop() {
  parseJson();
  delay(10000);
}
`

The serial monitor keeps printing parseObject() failed. I've tried changing the StaticJsonBuffer size, I've tried playing its declaration in various places, etc. but all to no avail. Help would be much appreciated.

bblanchon, 2016-05-25:
Hi Umar,

Please have a look at:

* [FAQ: The first parsing succeeds, why do the next ones fail? Cause 2: reuse of JSON input](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#cause-2-reuse-of-json-input).
* [Avoiding pitfalls: Make sure the string isn't read-only](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#7-make-sure-the-string-isnt-read-only)

bblanchon, 2016-05-25:
**TL;DR**: the `json` variable is modified by the first call to `parseObject()`. So only the first call can succeed, the next ones will fail.

uzbhutta, 2016-05-25:
Thanks for the reply. Much appreciated.

closed 2016-05-25 by uzbhutta

bblanchon, 2016-05-25:
You're welcome Umar, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: