bblanchon/ArduinoJson #287: Add install rules in CMakeLists.txt

mathieuhardy, 2016-05-26:
Hello,

I'd like to integrate your library as is in my project but when I perform a 'make install' the libArduinoJson.so is not installed. Is it possible to add in src/CMakeLists.txt an install rule ?

install(TARGETS  ArduinoJson  DESTINATION  lib).

Thank you very much.
Mathieu Hardy.

bblanchon, 2016-05-26:
Hi Mathieu,

I'm not sure it makes sense to add this now since ArduinoJson 6 will be a header only library.

However, if it's important to you, please make a PR.
But, to be exhaustive, you should also install the headers, don't you?

Regards,
Benoit

mathieuhardy, 2016-05-26:
Hi,

A header only library is a great news. So you're right, install rules are not usefull and we can close this issue. We'll use a workaround on our side for now. Do you know when this version will be released ?

Mathieu.

closed 2016-05-26 by mathieuhardy

bblanchon, 2016-05-26:
I don't know precisely, but I expect between two and four weeks from now.