bblanchon/ArduinoJson #288: Q: How can encoding float data being represented as scientific notation be stopped?

hemalchevli, 2016-05-28:
Background:
I'm using esp8266(with arduino framework on platformIO) and mega2560. Sensors are connected to mega which collects data, and that data is stored in json and sent to esp via UART, eps decodes this json, and encodes a new one, with some added parameters and sends it a webserver.
The following issue is on esp8266.
Input json is:
`{"DT":"28-05-2016 05:59:52","Meters":[3190.32,0.00,0.00,0.00,0.00,0.00,0.00,0.00],"BoxOpen":0}`
Output json is:
`{"DT":"28-05-2016 05:59:52","DeviceID":"Blk2","Meters":[3.19e3,0.00,0.00,0.00,0.00,0.00,0.00,0.00],"BoxOpen":0,"APIKey":"123456"}`
The first element of input json array in Meters is 3190.32, which in output is shown as scientific notation, which should be show as is in the input json as my webserver does not understand scientific notation.
Below is the function that does decoding and encoding

```
String prepareJSON(String jsondata) {
  const int BUFFER_SIZE = JSON_OBJECT_SIZE(12);
  StaticJsonBuffer<300> jsonBuffer; //for encoding final json to be sent
  StaticJsonBuffer<300> jsonBuf; //for decoding counters from mega
  String json = "";
  char counters[350];

  jsondata.toCharArray(counters, 350);// convert to char string from String
  //decoding
  JsonObject& countTime = jsonBuf.parseObject(counters);
  const char* dateTime  = countTime["DT"];
  int boxopen           = countTime["BoxOpen"];
  float c0             = countTime["Meters"][0];
  float c1             = countTime["Meters"][1];
  float c2             = countTime["Meters"][2];
  float c3             = countTime["Meters"][3];
  float c4             = countTime["Meters"][4];
  float c5             = countTime["Meters"][5];
  float c6             = countTime["Meters"][6];
  float c7             = countTime["Meters"][7];

  //encoding
  JsonObject& root = jsonBuffer.createObject();
  root["DT"] = dateTime;
  root["DeviceID"] = DevID;
  JsonArray& Meters = root.createNestedArray("Meters");
  //Meters.add(double_with_n_digits(c0, 2));
  Meters.add(c0,2);
  Meters.add(c1,2);
  Meters.add(c2,2);
  Meters.add(c3,2);
  Meters.add(c4,2);
  Meters.add(c5,2);
  Meters.add(c6,2);
  Meters.add(c7,2);
  root["BoxOpen"] = boxopen;
  root["APIKey"] = Key;

  root.printTo(json);
  jsondata = "";
  return json;
}
```
Can anyone please help me out here?

bblanchon, 2016-05-28:
Hi Hemal,

No, it's not possible to disable scientific notation.
As a side note, scientific notation is perfectly legal in JSON.
It offers a smaller representation and reduces the memory usage.

There is a workaround with the current version that consists in serializing the float in your code and feed it to ArduinoJson as a `RawJson` string.
But that's just a pain in the ass.
You'd better use `sprintf()` than go to this route.

Of course, I could release a new version of the library with a flag to disable scientific notation.
I'm not a big fan of this because it would increase the complexity of the library to work around a bug in your server. Moreover, I would need to double the number of tests for the float serialization.

hemalchevli, 2016-05-28:
I understand, thanks for your inputs, I think I'll use sprintf for now. After reading you reply I too don't see the point in adding a flag. I'll ask the web developer to fix the bug.
Thank you.

closed 2016-05-28 by hemalchevli

andrei-ivanov, 2017-03-09:
Btw, isn't there a bug if 100455.60 gets rounded up to 1.00e5 (100000)?

bblanchon, 2017-03-10:
Well not in a strict sense, you asked for two decimal places and it gives two decimal places.
If you want more digits, just increase the value.

See: 

* [JsonObject::set()](https://bblanchon.github.io/ArduinoJson/api/jsonobject/set/)

andrei-ivanov, 2017-03-10:
Hmm, but my number has 2 decimals:  100455.60

andrei-ivanov, 2017-05-20:
So... any way to specify 2 decimals now with 5.1.0? :)
The numbers seem to be serialized correctly, just with more decimals than I want.

bblanchon, 2017-05-21:
@andrei-ivanov, sorry there is no way to do that in 5.10.0.
If this is important for you, please [open a new issue](https://github.com/bblanchon/ArduinoJson/issues/new) and explain why you need this feature.