bblanchon/ArduinoJson #289: Adding a nested object to a parsed array

boraozgen, 2016-05-28:
Hello there,

First of all, amazing library! Instant star!

I read through the issues but could not find an answer to my question. Apologize me if it was asked before.

I am developing an ESP8266 application and using ArduinoJson library to save my configuration file. The application requires me to read and parse a JSON array of nested objects, and adding another nested object to the array before saving it again.

I've tried this:

`JsonObject& new_obj = json_config["NodeArray"].createNestedObject();`

where json_config is the parsed root object containing the nested array "NodeArray". This gives a compiler error since I am trying to assign a nested object to a subscript.

I have also thought to retrieve the array by assigning the NodeArray to a new array using .asArray(), adding the new object there, and afterwards adding it back to the root object. However, I think there is no way removing the array before adding the new one.

Is there a way to overcome this issue? What would you recommend?

Thank you

bblanchon, 2016-05-28:
Hi,

`json_config["NodeArray"]` returns a `JsonVariant` which doesn't have a `createNestedObject()` method.
To get this method, you need to cast the variant to a `JsonArray` or a `JsonObject`.
In your case:

```c++
JsonObject& new_obj = json_config["NodeArray"].as<JsonArray>().createNestedObject();
// or
JsonObject& new_obj = json_config["NodeArray"].asArray().createNestedObject();
// or
JsonArray& array = json_config["NodeArray"];
JsonObject& new_obj = array.createNestedObject();
```

That is basically what you described in the last paragraph, except that you seem to believe that a copy of the array is made. But that's not the case: `asArray()` returns a reference to the original array, so the changes are done in the root object too.

Regards,

boraozgen, 2016-05-28:
Hi Benoit,

Thank you for your swift answer. All is clear now :)

All the best

boraozgen, 2016-05-28:
Hi,

The code does not seem to work. Here is my expanded code:

```
// If nodeArray exists, add new object
      if (json_config["nodeArray"].is<JsonArray&>())
      {
        Serial.println("nodeArray exists. Adding a new node.");
        JsonObject& new_obj = json_config["NodeArray"].asArray().createNestedObject();
        new_obj["nodeAddress"] = new_dev_addr;
        new_obj["nodeType"] = "DHT";
        new_obj["txInterval"] = 5;
      }
      // If nodeArray does not exist, create array and add the first element
      else
      {
        Serial.println("No nodeArray found. Creating a new one.");
        JsonArray& nodeArray = json_config.createNestedArray("nodeArray");
        // Add new node to nodeArray
        JsonObject& new_obj = nodeArray.createNestedObject();
        new_obj["nodeAddress"] = new_dev_addr;
        new_obj["nodeType"] = "DHT";
        new_obj["txInterval"] = 5;
      }
```

The "else" part works perfectly, creating an array and adding the object without any problems.
However the "if" part does not change anything in the json_config object (the serial print line is printed, so the program enters the if statement). The printed JSON string stays like this (nodeCount is handled in another part of the code):

`{"nodeCount":2,"nodeArray":[{"nodeAddress":1,"nodeType":"DHT","txInterval":5}]}`

whereas it should have been:

`{"nodeCount":2,"nodeArray":[{"nodeAddress":1,"nodeType":"DHT","txInterval":5},{"nodeAddress":2,"nodeType":"DHT","txInterval":5}]}`

What might be the problem?

Thanks

bblanchon, 2016-05-29:
Hi,

I wrote [a test](https://github.com/bblanchon/ArduinoJson/blob/dd025f7804771842e14ea12543ffcf23f495c4a7/test/Issue289.cpp) to reproduce your issue but it works as expected.

My guess is that the `JsonBuffer` is too small.

See:

* [Avoiding pitfalls: Make StaticJsonBuffer big enough](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#1-make-staticjsonbuffer-big-enough)
* [FAQ: What are the differences between StaticJsonBuffer and DynamicJsonBuffer?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer)
* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#what-are-the-differences-between-staticjsonbuffer-and-dynamicjsonbuffer)
* [FAQ: The first parsing succeeds, why do the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-parsing-succeeds-why-do-the-next-ones-fail)

Regards.

boraozgen, 2016-05-30:
Dumb me. It appears to be a simple typo. As you can see, I wrote down "NodeArray" instead of "nodeArray" in this line:

`JsonObject& new_obj = json_config["NodeArray"].asArray().createNestedObject();`

It works now since I copied the line from your example.

Thanks

bblanchon, 2016-05-30:
No biggie :wink:

closed 2016-05-30 by bblanchon