bblanchon/ArduinoJson #290: Use of Deleted Function - Using JsonObject as member variable

bepursuant, 2016-05-29:
Hello - I threw a small library on top of this to allow saving/loading configuration from a json file, and I was hoping to be able to expose a JsonObject through my config class so I could just interact with it as config.jsonObj["value"]... I was able to get the class working with the Buffer and Object defined within the init() and save() functions, but when I try to pull the Buffer and Object into member variables, it fails with the below errors.

- use of deleted function 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)'
- 'ArduinoJson::JsonObject& ArduinoJson::JsonObject::operator=(const ArduinoJson::JsonObject&)' is implicitly deleted because the default definition would be ill-formed
- 'ArduinoJson::Internals::ReferenceType& ArduinoJson::Internals::ReferenceType::operator=(const ArduinoJson::Internals::ReferenceType&)' is private

I have the whole code posted in a gist here https://gist.github.com/bepursuant/fc558c18f409271583fe3d3331fe8b5a , but the offending line is:
```
this->jsonObj = this->jsonBuf.parseObject(buf.get());
```

I'm sure my question stems from a misunderstanding of the buffers/scoping in C++, or of exactly how this library should be used... so I apologize upfront for the probably somewhat ignorant question.

bblanchon, 2016-05-29:
Hi,

It's a rule of the C++ language: you cannot reassign a reference.
As a consequence, if you have a member of type reference, you have to set it in the member initializer list. Unfortunately, this is not possible in your case.

The solution is to use a pointer instead of a reference:

```c++
class ConfigFile{
  public:
    JsonObject& jsonObject() {
        return *jsonObjectPtr;
    }

  private:
    JsonObject* jsonObjPtr;
};
```

However, I think I would be more clever to use a `JsonVariant`, as it would wrap the pointer for you, and would also support arrays:

```c++
class ConfigFile{
  public:
    JsonObject& jsonObject() {
        return jsonVariant;
    }

  private:
    JsonVariant jsonVariant;
};
```

Regards.

bblanchon, 2016-06-04:
@bepursuant Any progress? Can I close the issue?

bepursuant, 2016-06-05:
@bblanchon Yes, as usual your library has been a fantastic help, but you've been fantastic supporting it as well! Again, apologies for the semi-unrelated-to-your-library question, but I truly appreciate your willingness to help anyway :) I was able to get my library working using the JsonVariant method you mentioned, which worked with almost no additional modifications.

closed 2016-06-05 by bepursuant

ccostel, 2017-04-30:
@bepursuant I am trying to do something similar. Would you be willing to share the gist for the changed code?