bblanchon/ArduinoJson #293: Doesn't work with multiple items

coollorenzo, 2016-05-29:
The libary works great with json data with one item `{"bericht":"is al aangemeld<br>"}
` `StaticJsonBuffer<200> jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(response);
    String bericht = root["bericht"];` returns "is al aangemeld<br>" as it should but when i trie it with: `{"id":"2","voornaam":"lorenzo","achternaam":"nannings","geboortedatum":"1990-09-19","email":"example@email.com","telefoonnummer":"0612345678","aanmeldingsdatum":"2014-01-07 00:00:00","laatsteles":"8ee5f57tp8vg6d0f13d72cudvd","tegoed":"4294967234","bericht":"lorenzo nannings<br>tegoed: 4294967234"}` I get nothing is it user error or a bug in the libary?

bblanchon, 2016-05-29:
Hi Lorenzo,

I believe that the `StaticJsonBuffer` is too small: 200 bytes is probably not enough.

Please have a look at:

* [User Guide: Decoding JSON](https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON)
* [Avoiding pitfalls: Make StaticJsonBuffer big enough](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding-pitfalls#1-make-staticjsonbuffer-big-enough)
* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

Regards.

coollorenzo, 2016-05-29:
Thanks I forgot about that now it works great again!

bblanchon, 2016-05-29:
You're welcome, Lorenzo.
Thank you for using ArduinoJson.

closed 2016-05-29 by bblanchon