bblanchon/ArduinoJson #294: Publish the Library for Particle devices

hl68fx, 2016-05-29:
Hi,

it would be cool if you could publish your library for Particle devices like the Core, Photon and Electron.
[www.particle.io](www.particle.io)
[Here](https://github.com/spark/uber-library-example) is a guide how to publish a library for [build.particle.io](build.particle.io)

Kind regards,
Tom

bblanchon, 2016-05-30:
Hi Tom, 

In 2014, I wanted to publish it for "Spark", but unfortunately, they impose an unusual library layout.
[An issue on this point](https://github.com/spark/uber-library-example/issues/3) has been opened for two years, and I don't think it's fixed yet.

So I guess someone else publish the library for me.
To this person: thank you :+1:

Regards and thanks for using ArduinoJson.

kenvernaillenvub, 2017-02-21:
any progress on this?

bblanchon, 2017-02-22:
Apparently, they updated the library format, I'll see what I can do.
Any help is welcome.

foodbag, 2017-03-05:
I was able to get the 5.8.2 release of ArduinoJson to compile with the new lib format. It works as an library uploaded to the particle servers and needed only minimal changes (no need to flatten the library structure anymore).  I'll try to generate a quick fork to show the needed modifications.

bblanchon, 2017-03-07:
See #461

bblanchon, 2017-04-24:
[ArduinoJson 5.9.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.9.0) is finally compatible with Particle.

closed 2017-04-24 by bblanchon