bblanchon/ArduinoJson #295: how to parse JSON objects inclosed in a JSON array which is further inclosed in a JSON object?

HasnainKanani, 2016-05-30:
here is the sample:
{
	"scientists": [{
		"name": "Euler",
		"x": 1,
		"y": 2,
		"orientation": "E"
	}, {
		"name": "Omar Khayyam",
		"x": 4,
		"y": 5,
		"orientation": "S"
	}],
	"safezones": [{
		"scientist": "Euler",
		"x": 3,
		"y": 5,
		"orientation": "W"
	}, {
		"scientist": "Omar Khayyam",
		"x": 4.5,
		"y": 5,
		"orientation": ""
	}]
}

bblanchon, 2016-05-30:
Here is the solution:

```c++
JsonObject& root = jsonBuffer.parseObject(json);
const char* name0 = root["scientists"][0]["name"];
```

@HasnainKanani, I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

hemantsangwan, 2016-05-30:
@HasnainKanani refer to https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON for more details. Most of the topics are already covered in wiki

closed 2016-06-04 by bblanchon