bblanchon/ArduinoJson #296: problème et parse ...

eqcq, 2016-06-01:
Bonjour et merci pour cette très bonne librairie.
j'ai eu déjà la chance de pouvoir l'utiliser dans quelques projets.
Cependant j'ai un léger problème de parsing avec le petit dernier. 
je n'arrive pas a determiner le type de variable contenu dans mon json. En effet dans le moniteur série mes variables ne s'affichent pas, bien que j'arrive a afficher le json au complet. J'en déduis, si je me fie au wiki, que mes variables ne doivent pas être de la bonne catégorie

voici un extrait du code (payload est issu d'un websocket) :

String result=String((const char *)payload);
int size = result.length() + 1;
 char json[size];
 result.toCharArray(json, size);
 USE_SERIAL.println(result);
StaticJsonBuffer<200> jsonBuffer;
 JsonObject& json_parsed = jsonBuffer.parseObject(json);      
const char*  action=json_parsed["action"];
const char * magnitude=((const char *)json_parsed["data"]["properties"]["mag"]);
 USE_SERIAL.println(action);
 USE_SERIAL.println(magnitude);

et voici un exemple de json reçu :

{"action":"create","data":{
  "geometry": {
    "type": "Point", 
    "coordinates": [
      -67.06, 
      -23.57, 
      -235.0
    ]
  }, 
  "type": "Feature", 
  "id": "20160601_0000005", 
  "properties": {
    "lastupdate": "2016-06-01T03:18:00.0Z", 
    "magtype": "ml", 
    "evtype": "ke", 
    "lon": -67.06, 
    "auth": "GUC", 
    "lat": -23.57, 
    "depth": 235.0, 
    "unid": "20160601_0000005", 
    "mag": 3.7, 
    "time": "2016-06-01T02:58:58.0Z", 
    "source_id": "509728", 
    "source_catalog": "EMSC-RTS", 
    "flynn_region": "JUJUY, ARGENTINA"
  }
}}

j'ai consulté le wiki et fait plusieurs essais mais pour l'instant en vain. Si tu pouvais m'éclairer ce serait très apprécié merci.

bblanchon, 2016-06-01:
Hi Guillaume
*Bonjour Guillaume,*

I think the `StaticJsonBuffer` is too small.
*Je pense que le `StaticJsonBuffer` est trop petit.*

Please have a look at:
*Jette un coup d'oeil à:*

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* [ArduinoJson: Memory Model](https://github.com/bblanchon/ArduinoJson/wiki/Memory%20model)
* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)

Regards,
*Cordialement,*

Benoit

eqcq, 2016-06-01:
Thanks a lot Benoit, it was exactly that, I've increase the size of the buffer and everything is ok now. 👍 
best regards

bblanchon, 2016-06-01:
You're welcome.
Thank you for using ArduinoJson.

closed 2016-06-01 by bblanchon