bblanchon/ArduinoJson #297: reading json file from SD card failed

umer427, 2016-06-01:
I need to read a json file in .json extension from my SD card mounted on Arduino Mega from the help of http://www.comm2excel.com/projects/sd_card_module.php?id=15# 

SD card is loading pretty well but when i try to read json file, it doesnt, please help me out

`void readJSON() {

  File dataFile;
  
   dataFile= SD.open("amm.json", FILE_READ);   
  String json = dataFile.readString();                    
  dataFile.close();                                      

  StaticJsonBuffer<4000> jsonBuffer;      
  JsonObject& root = jsonBuffer.parseObject(json);    

  Serial.println(json);

  if (!root.success()) {
    Serial.println("parseObject() failed");
    return;   

  }
  else { // update variables i will write my arays over here, dont worry about it

   Serial.println("hi");
    
  }
}`

Serial monitor :  
`Initializing SD card...initialization done.
done!

parseObject() failed

`

thank you.

bblanchon, 2016-06-01:
It heavily depends on the content of the file.
My bet is that the string `json` is pretty big and that it doesn't fit in the `StaticJsonBuffer`.
The solution, if any, is to use `readBytes()` instead of `readString()`.

See:

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)

umer427, 2016-06-01:
Thank you for your quick reply. Actually I forgot to appreciate your great work Sir. I will definitely like to praise you for it. You have done really a  great work. 
Actually i have the same question as someone asked it before.
I want to parse it, by using enough dynamic buffer size, it doesn't works for me. I know i am definitely wrong. Please help me out in it. Thank you 

> {
> "scientists":[
> {
> "name":"Euler",
> "x":1,
> "y":2,
> "orientation":”E” },
> {
> "name":"Omar Khayyam",
> "x":4,
> "y":5,
> "orientation":”S”
> }
> ],
> "safezones":[
> {
> "scientist":"Euler",
> "x":3,
> "y":5,
> "orientation":”W”
> },
> {
> "scientist":"Omar Khayyam",
> "x":4.5,
> "y":5,
> "orientation":””
> }
> ]
> }

bblanchon, 2016-06-01:
I'd be glad to help you :bowtie:, but I need further information.

* What did you try?
* What result did you expect?
* What happened instead?

Please provide a detailed description of the problem.

bblanchon, 2016-06-01:
:open_mouth:
http://www.nerc.com.pk/Categories/Indigenous

bblanchon, 2016-06-01:
LOL :smile:

umer427, 2016-06-01:
Yeah it is an example from  nerc. Regardless NERC. I am here to learn just json. NERC categories are required to design robots & there path planning algorithms. But am not doing that. Its just an example of nested arrays, which i found preferred for a good example of parsing. If you have a better example, please give me that one. Thank you

umer427, 2016-06-02:
No response?

bblanchon, 2016-06-02:
My answer was

> I'd be glad to help you :bowtie:, but I need further information.
> 
> * What did you try?
> * What result did you expect?
> * What happened instead?
> 
> Please provide a detailed description of the problem.

See above.

BTW, please realize that, like most open-source contributors, I do this on my own time, and I do this for free. Time I spend solving your problem is time I don't spend with my kids.

Sentences like "No response?" are totally inappropriate :angry:. I could give you no answer at all, and that would be 100% cool. In fact, that what's happens on many projects, and that's OK because people are busy with other things and we are thankful for what they already gave.

umer427, 2016-06-03:
I am sorry about it. 
actually I am facing the problem that parseobject keep failing, as i have used static buffer & dynamic buffer both individually. may there's something wrong in the method of my writing code.

>  File dataFile;
>   
>    dataFile= SD.open("ma.json", FILE_READ);   //open file (path has been set elsewhere and works)
>   String json = dataFile.readString();                    // read data to 'json' variable
>   dataFile.close();            
> 
> DynamicJsonBuffer jsonBuffer;
> 
> JsonObject& root = jsonBuffer.parseObject(json);    // breaks the JSON string into individual items, saves into JsonObject 'root'
>  const char* g = root["scientists"][0]["name"].asString();
> 
>   Serial.println(g);
> 
>   if (!root.success()) {
>     Serial.println("parseObject() failed");
>     return;   //if parsing failed, this will exit the function and not change the values to 00
> 
>   }
>   else { // update variables
> 
>    Serial.println("hi");
>     
>   } 
> 

also i have used the technique of getting size of our json, but still fails

> 
> const int sizee=JSON_OBJECT_SIZE(2) + JSON_ARRAY_SIZE(32);
> StaticJsonBuffer<sizee> jsonBuffer;
> 
> JsonObject& root = jsonBuffer.parseObject(json);    // breaks the JSON string into individual items, saves into JsonObject 'root'
>  const char* g = root["scientists"][0]["name"].asString();
> 
>   Serial.println(g);

I know it's pretty hard to get spare time for people like me & help them without any honorarium. I am very obliged to you for your efforts.

hemantsangwan, 2016-06-03:
Hi @umer427 , I think your answer to question https://github.com/bblanchon/ArduinoJson/issues/297#issuecomment-223021143 is still incomplete.

umer427, 2016-06-03:
@hemantsangwan  I have updated my comment again.

bblanchon, 2016-06-03:
OMFG, I think I found the problem!!!!!!!!

[image]

Unbelievable!
I guess that's what happens when you write JSON files in Word :smile: 

I'm sorry @umer427, ArduinoJson cannot work if the input is not proper JSON.

bblanchon, 2016-06-03:
For the record, you could have spotted the issue using any kind of editor with syntax highlighting.

For example, if I copy/paste the text in GitHub, we can clearly see the quotation marks in red.

```json
                                            {  

   "scientists":[  

      {  

         "name":"Euler",

         "x":1,

         "y":2,

         "orientation":”E”

      },

      {  

         "name":"Omar Khayyam",

         "x":4,

         "y":5,

         "orientation":”S”

      }

   ],

   "safezones":[  

      {  

         "scientist":"Euler",

         "x":3,

         "y":5,

         "orientation":”W”

      },

      {  

         "scientist":"Omar Khayyam",

         "x":4.5,

         "y":5,

         "orientation":””

      }

   ]

} 

```

umer427, 2016-06-03:
Yeah the online files are in wrong syntax. I know these mistakes. I corrected them using notepad++ 
I have the correct files too. 
I mentioned NERC just for an example to understand parsing. Please help me out. As i always gets parsing failed even on dynamic buffer. Please

bblanchon, 2016-06-03:
Did you try to replace `readString()` by `readBytes()` as I suggested in my first message?
Did you check each recommendation of [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails), as I suggested in my first message?

bblanchon, 2016-06-03:
Also, it could help if you tell us what is the actual content of the file... and of course what hardware you're using...

umer427, 2016-06-03:
@bblanchon  Yeah i have replaced by readBytes() but what should be the data type of json?
I assumed it as byte too but still it doesn't compile.
 

> File dataFile;
>    dataFile= SD.open("ma.json", FILE_READ);   //open file (path has been set elsewhere and works)
>   String json = dataFile.readBytes();                    // read data to 'json' variable
>   dataFile.close();    

Error: exit status 1
no matching function for call to 'SDLib::File::readBytes()'

umer427, 2016-06-03:
I am using Arduino Mega with SD card module. Module is working good as i can read Txt files easily from it.
i am in serious need please. also can you brief me how to make donations, give the link please

umer427, 2016-06-03:
here is what i am reading simple example of 

{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}

saved in SD card as .JSON extension using notepad++ 

but still parsing fails. i think i am noob. :-( 

```

> #include <SPI.h>
> #include <SD.h>
> #include <ArduinoJson.h>
> void setup() {
>  
>   Serial.begin(9600);
>    while (!Serial) {
>     ; // wait for serial port to connect. Needed for Leonardo only
>   }
> 
> 
>   Serial.print("Initializing SD card...");
>   pinMode(53, OUTPUT);
> 
>   if (!SD.begin(53)) {
>     Serial.println("initialization failed!");
>     return;
>   }
>   Serial.println("initialization done.");
> 
> 
>   Serial.println("done!");
> 
> 
> 
> 
>  File dataFile;
>   
>    dataFile= SD.open("gg.json", FILE_READ);   //open file (path has been set elsewhere and works)
>   String json = dataFile.readString();                    // read data to 'json' variable
>   dataFile.close();            
> 
> 
> StaticJsonBuffer<400> jsonBuffer;
> 
> JsonObject& root = jsonBuffer.parseObject(json);  
> //char json[] = "{\"sensor\":\"gps\",\"time\":1351824120,\"data\":[48.756080,2.302038]}";
> 
> 
> if (!root.success())
> {
>   Serial.println("parseObject() failed");
>   return;
> }
> 
> //
> // Step 3: Retrieve the values
> //
> const char* sensor    = root["sensor"];
> long        time      = root["time"];
> double      latitude  = root["data"][0];
> double      longitude = root["data"][1];
> 
> }

```

bblanchon, 2016-06-03:
The [JsonHttpClient example](https://github.com/bblanchon/ArduinoJson/blob/master/examples/JsonHttpClient/JsonHttpClient.ino#L115) shows how to use `readBytes()`.

Please try:

```c++
// ...unchanged...

#define MAX_FILE_SIZE 1024
#define JSON_BUFFER_SIZE 200

void readFile(const char *fileName, char *buffer, size_t maxSize) {
  File file = SD.open(fileName, FILE_READ);
  file.readBytes(buffer, maxSize);
  file.close();
}

void setup() {
  // ...unchanged...

  char json[MAX_FILE_SIZE];
  readFile("gg.json", json, MAX_FILE_SIZE);

  Serial.println(json); // <- SHOW ME THE OUTPUT OF THIS!!!!!

  StaticJsonBuffer<JSON_BUFFER_SIZE> jsonBuffer;

  // ...unchanged...
}
```

umer427, 2016-06-03:
@bblanchon  I understand now. The output is : 

```
Initializing SD card...initialization done.
done!
Ü´;ß*Þb½~'ßëŸ_ûž=ûü_ß¥óþíÿ,«æ>7ÿù7©øKÖ™½eÿ¾{¾-¿÷{½Úû¿öŸÓî÷ïï�üŸãìþ;÷ûóÜ¿z÷qz5³ô?ûÎéíþß¸ÿ~Ž_óÿôöôó¾½ß³»ÿ~î¿Ý½ïýP	Õþ¯ù½òãïòêœµ>ûÿÿÉö«¾u÷Íézw»÷yñJ¶¿öó·5ÿÛk

```

umer427, 2016-06-03:
Now its reading good but in some other formats. What to do now?
Like
Ü´;ß*Þb½�~'ßëŸ_ûž=�ûü_ß¥óþíÿ,«æ>7ÿù7©øKÖ™½�eÿ¾{¾-¿÷{½Úû¿öŸÓî÷ïï�üŸãìþ;÷ûóÜ¿z÷qz5³ô?ûÎéíþß¸ÿ~Ž�_óÿôöôó¾�½ß³»ÿ~î¿Ý½ïýP  Õ�þ¯ù½òãïòêœµ>ûÿÿÉö«¾u÷�Íézw»÷yñJ�¶¿öó·5ÿÛ�k

hemantsangwan, 2016-06-04:
@umer427 In context of https://github.com/bblanchon/ArduinoJson/issues/297#issuecomment-223525246 refer to main page or directly click [here](https://www.paypal.com/cgi-bin/webscr?cmd=_donations&business=donate%40benoitblanchon%2efr&lc=GB&item_name=Benoit%20Blanchon&item_number=Arduino%20JSON&currency_code=EUR&bn=PP%2dDonationsBF%3abtn_donate_LG%2egif%3aNonHosted). :wink: 

I may sound silly but sometimes I forget to match baud rate, parity etc... I hope that's not the reason :laughing:

umer427, 2016-06-04:
Alright. Thanks for the link.  Please resolve my problem

bblanchon, 2016-06-04:
@umer427 Your issue is not related to ArduinoJson; here is not the right place to ask for help.

The only advice I can give you is to start with the examples provided with the SD library.
Also, have a look at the [documentation for SD library](https://www.arduino.cc/en/Reference/SD).

closed 2016-06-04 by bblanchon