bblanchon/ArduinoJson #299: Error when using the library ArduinoJson Intel Galileo

icaroramiires, 2016-06-01:
When I compile the project for the Arduino uno works perfectly.

I am now moving my project for intel and Galilean appeared the following error:

In file included from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonWriter.hpp:10:0,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:13,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:13,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:15,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /Users/icaro/Documents/projeto_final/projeto_final.ino:1:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp: In function ‘bool ArduinoJson::Polyfills::isInfinity(T) [with T = double]’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp:35:18: error: ‘isinfl’ was not declared in this scope
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp: In function ‘bool ArduinoJson::Polyfills::isInfinity(T) [with T = float]’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp:40:18: error: ‘isinff’ was not declared in this scope
In file included from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonWriter.hpp:11:0,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:13,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:13,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:15,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /Users/icaro/Documents/projeto_final/projeto_final.ino:1:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp: In function ‘bool ArduinoJson::Polyfills::isNaN(T) [with T = double]’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp:35:18: error: ‘isnanl’ was not declared in this scope
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp: In function ‘bool ArduinoJson::Polyfills::isNaN(T) [with T = float]’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp:40:18: error: ‘isnanf’ was not declared in this scope
In file included from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariantBase.hpp:11:0,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:16,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:15,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /Users/icaro/Documents/projeto_final/projeto_final.ino:1:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonObjectKey.hpp: In constructor ‘ArduinoJson::JsonObjectKey::JsonObjectKey(const String&)’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonObjectKey.hpp:18:49: error: ‘const class String’ has no member named ‘c_str’
In file included from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10:0,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /Users/icaro/Documents/projeto_final/projeto_final.ino:1:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function ‘ArduinoJson::JsonArray& ArduinoJson::JsonBuffer::parseArray(const String&, uint8_t)’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:71:28: error: ‘const class String’ has no member named ‘c_str’
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function ‘ArduinoJson::JsonObject& ArduinoJson::JsonBuffer::parseObject(const String&, uint8_t)’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:94:29: error: ‘const class String’ has no member named ‘c_str’
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function ‘ArduinoJson::JsonVariant ArduinoJson::JsonBuffer::parse(const String&, uint8_t)’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:109:23: error: ‘const class String’ has no member named ‘c_str’
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp: In member function ‘char* ArduinoJson::JsonBuffer::strdup(const String&)’:
/Users/icaro/Documents/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:116:55: error: ‘const class String’ has no member named ‘c_str’
exit status 1

Error compiling for Intel Galileo board

Version IDE = 1.6.9
Version lib = 5.5.0 
Mac OSX El Capitan

bblanchon, 2016-06-01:
It's a bug in galileo corelibs.
I submitted a [Pull Request](https://github.com/01org/corelibs-galileo/pull/29)

I also updated the [Wiki](https://github.com/bblanchon/ArduinoJson/wiki/Compatibility-issues#intel-galileo-compilation-error)

Thank you very much for reporting this problem.

bblanchon, 2016-06-04:
The PR has been merged :tada:

igoromero, 2016-06-05:
@bblanchon, I'm working on the same project. Getting to ArduinoJson run on Intel Galileo thanks to Wiki suggestion. However has an error in the designated directory.

The directory of WString.h file is correct for the Intel Edson board:
% LOCALAPPDATA% \ Arduino15 \ packages \ Intel \ hardware \ i686 \ 1.6.2 + 1.0 \ colors \ Arduino \ WString.h

However, Intel Galileo Gen1 and Gen 2 card uses i586 that is the directory for this card is in

in Windows
% LOCALAPPDATA% \ Arduino15 \ packages \ Intel \ hardware \ i586 \ 1.6.2 + 1.0 \ colors \ Arduino \ WString.h

on mac

~ / Library / Arduino15 / packages / Intel / hardware / i586 / 1.6.7 + 1.0 / colors / arduino / WString.h

Another detail that influenced was the version of ArduinoJson library, so get to work in version 5.0.0.

bblanchon, 2016-06-05:
Thanks @igoromero.
I updated the wiki.
Next time, do not hesitate to edit yourself :wink:

muhammadnizami, 2016-06-06:
I followed the wiki, added the c_str function.

The error message on c_str function disappeared, but I still have problems with ArduinoJson::Polyfills on Intel Galileo. It compiles for Arduino Uno but not for Intel Galileo.

In file included from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonWriter.hpp:10:0,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:13,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:13,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:15,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:8:
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp: In function 'bool ArduinoJson::Polyfills::isInfinity(T) [with T = double]':
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp:35:18: error: 'isinfl' was not declared in this scope
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp: In function 'bool ArduinoJson::Polyfills::isInfinity(T) [with T = float]':
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isInfinity.hpp:40:18: error: 'isinff' was not declared in this scope
In file included from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonWriter.hpp:11:0,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/JsonPrintable.hpp:13,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonVariant.hpp:13,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../JsonBuffer.hpp:15,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/BlockJsonBuffer.hpp:10,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/DynamicJsonBuffer.hpp:10,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson.h:8,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/src/ArduinoJson.h:14,
                 from /home/nim_13512501/Arduino/libraries/ArduinoJson/examples/JsonGeneratorExample/JsonGeneratorExample.ino:8:
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp: In function 'bool ArduinoJson::Polyfills::isNaN(T) [with T = double]':
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp:35:18: error: 'isnanl' was not declared in this scope
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp: In function 'bool ArduinoJson::Polyfills::isNaN(T) [with T = float]':
/home/nim_13512501/Arduino/libraries/ArduinoJson/src/../include/ArduinoJson/Internals/../Internals/../Polyfills/isNaN.hpp:40:18: error: 'isnanf' was not declared in this scope

bblanchon, 2016-06-06:
@muhammadnizami I didn't have time to try it yet. Any chance this could be related to #284?

muhammadnizami, 2016-06-06:
@bblanchon I will look at it when I have time.

igoromero, 2016-06-06:
@muhammadnizami Which version of the library that you are using? Try using version 5.1.1

muhammadnizami, 2016-06-06:
@igoromero I had used 5.5.0 when I had the Issue.

Thanks. I tried 5.1.1 and it compiles.

bblanchon, 2016-06-06:
I fixed the issue in v5.5.1 :heavy_check_mark: 
Thank you for reporting this.

closed 2016-06-06 by bblanchon