bblanchon/ArduinoJson #300: Support for Stream as input

bblanchon, 2016-06-04:
Copy the content of the stream in the `JsonBuffer`
Ideally, remove the spaces to use less memory.

Diaoul, 2016-10-11:
I'm reading JSON from Serial and I'd like to avoid an extra buffer. Would that work for me?

bblanchon, 2016-12-22:
@Diaoul Yes, the input will be copied in the `JsonBuffer` automatically.

bblanchon, 2017-01-03:
Feature available in release [5.8.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v5.8.0)

closed 2017-01-03 by bblanchon

sticilface, 2017-01-10:
Double thumbs up for this!   big help :)  thanks!!!

bblanchon, 2017-01-12:
Thanks @sticilface, it's been quite a challenge.

It's much more efficient than using `Stream::readBytes()` as before as the `JsonBuffer` will duplicate the minimal amount of data, thus saving a lot of memory if the input JSON is indented.

The [ArduinoJson Assistant](https://bblanchon.github.io/ArduinoJson/) has been updated to tell the number of bytes that are required for the duplication.