bblanchon/ArduinoJson #301: printTo wifi Client too slow

domingosl, 2016-06-05:
Is there any reason why printTo is too slow? I did some experiments sending some json string to my socket server and if you use:

```
    client.write("{\"cmd\":1,\"chipId\":\"15952839\",\"sign\":\"670c4560128af4691bab40408751f4fd857f6ea49ef804b4de6e7e77e741c587\",\"nounce\":404,\"payload\":\"ciao mondo!\"}");
    client.write("\n", 1);
```

It takes around 50ms to send everything (Im using a ESP8266), and all the data get send in a single TCP packet. Sending the same JSON data but using:

```
    root.printTo(client);
    client.write("\n", 1);
```

Takes over 6 seconds, and each char is send in a separate TCP packet.

Is there a way to replicate the behavior of the first example using printTo? 

BTW, im using nodejs to mesure the time like this:

   ```
 socket.on('data', function (data) {
        try {
            //console.log(data);
            //data = JSON.parse(data);

            if(!timerStarted) {
                sTime = new Date().getTime();
                timerStarted = true;
            }

            if (data == "\n") {
                console.log((buffer.toString()));
                buffer = new Buffer(0, 'hex');
                var end = new Date().getTime();
                var time = end - sTime;
                console.log("Elapsed time: " + time);
            }
            else
                buffer = Buffer.concat([buffer, new Buffer(data, 'hex')]);

            //command.exec(socket, data.cmd, data.payload);
        }
        catch (e) {
            socket.end();
            log.warn(e);
        }

    });
```

And the result is:

```
{"cmd":1,"chipId":"15952839","sign":"670c4560128af4691bab40408751f4fd857f6ea49ef804b4de6e7e77e741c587","nounce":404,"payload":"ciao mondo!"}
Elapsed time: 46
{"cmd":1,"chipId":"15952839","sign":"7cc46175e0374d50485a1fbd1fcd3e559d1949e51975acd356e1c85a7bd7728a","nounce":12,"payload":"ciao mondo!"}
Elapsed time: 6512

```

bblanchon, 2016-06-06:
I added my answer to the FAQ:

* [FAQ: Why ArduinoJson is slow?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-arduinojson-is-slow)

domingosl, 2016-06-06:
Thank you!, I'm going to use the proxy solution. BTW, you forgot a "_" in buffer[_size] = '\0'; right after the flush function declaration.

bblanchon, 2016-06-06:
It's a wiki! Please fix the page :wink:

bblanchon, 2016-06-08:
@domingosl Did that solve your speed problem? Can I close this issue?

domingosl, 2016-06-09:
@bblanchon Yes, thank you.

bblanchon, 2016-06-09:
You're welcome, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-06-09 by bblanchon