bblanchon/ArduinoJson #302: How to calculate the buffer size of an object with an array of arrays?

feklee, 2016-06-08:
Example:

    {"type":"scan data","data":[[0,0,0],[0,1,0],[1,0,0],[1,1,0]]}

bblanchon, 2016-06-08:
[image]

Provided that you're using `char*` and not `String`, the required size:

JSON_OBJECT_SIZE(2)+JSON_ARRAY_SIZE(4)+4*JSON_ARRAY_SIZE(3)

Make sure to have a look at:

* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)

feklee, 2016-06-09:
Thanks for clarification! Actually, what you propose is what I did before, and it didn't work until I added some padding. It turned out that it was a memory management issue, caused by me not paying attention.

bblanchon, 2016-06-09:
No biggie. 
I'm sure that picture will be helpful for others.
Thanks for using ArduinoJson.

closed 2016-06-09 by bblanchon