bblanchon/ArduinoJson #303: Conversion from 'ArduinoJson::JsonVariant' to (any enum) is ambiguous

uzbhutta, 2016-06-13:
Hi there,

I'm having trouble with populating parsed content into enums using this library. Consider the following:

`enum FLAG {GET, POST, PUT, DELETE};

typedef struct 
{
  FLAG flag;
  int time field1;
} sample_t
`

In main, after calling jsonBuffer.parseObject,
`
sample_t sample;

sample.field1 = row["field1"];
sample.flag = row["flag"];
`

However, while the field1 = row["field1"] works fine, the next line gives the error:
Conversion from 'ArduinoJson::JsonVariant' to 'FLAG' is ambiguous

I've tried using instead an explicit conversion:
`sample.flag = row["flag"].as<int>();`

However, this gives the error
Invalid conversion from 'int' to 'FLAG'

Any idea on how I can keep the enum design and integrate it with the JSON library? Thanks a lot as always.

bblanchon, 2016-06-13:
Hi Umar,

It's an interesting point where C and C++ differ.

* In C, you can implicitly cast an integer to an enum ([demo](https://ideone.com/Ws2C9y))
* In C++, you need an explicit cast ([demo](https://ideone.com/RRueCA))

So the solution to your issue is to explicitly cast the integer:

```c++
sample.flag = (FLAG)row["flag"].as<int>();
// or
sample.flag = static_cast<FLAG>(row["flag"].as<int>());
```

With a bit of template metaprogramming, I beleive it's possible to implement something lile

```c++
sample.flag = row["flag"].as<FLAG>();
```

...but I don't think it's a good idea because users would believe that this works with strings too ("GET", "POST"...).

uzbhutta, 2016-06-13:
Thanks so much for your prompt reply as always.
Worked like a charm!

closed 2016-06-13 by uzbhutta