bblanchon/ArduinoJson #304: fail to handle escaped binary data in json string

soubok, 2016-06-14:
Hello,
I try to decode json with binary data in a string:

```
  char json[] = "{\"data\":\"\\u0001\"}";
  StaticJsonBuffer<50> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(json);
  String str(root["data"].as<String>());
  for ( int i = 0; i < str.length(); ++i )
    Serial.println((byte)str.charAt(i));
```

This code prints "117 48 48 48 49" (u 0 0 0 1) but I expect "1"

bblanchon, 2016-06-14:
Hello,

Indeed, the escape sequence `\u` is not supported and I don't think it will ever be.
Converting a UTF-16 surrogate pair to UTF-8 can be quite a challenge, and this is very likely to make the code size blow.

The current implementation is supposed to leave the sequence unchanged, so the output should be "\u0001" and not "u0001". This is a bug.

soubok, 2016-06-14:
Maybe a compromise would be to support only 8 bits to have at least the following behavior:
```
  JSON.parse("{\"data\":\"\\u0001\"}").data.charCodeAt(0) -> 1
  JSON.parse("{\"data\":\"\\u00FF\"}").data.charCodeAt(0) -> 255
```

bblanchon, 2016-06-14:
But in that case, ArduinoJson would not be able to print the original string back, which is a very common use case.
We need to solve both the deserialization and the serialization.

soubok, 2016-06-14:
that's true, thanks for your help !

soubok, 2016-06-14:
Yes, I am very happy with ArduinoJson, and I will just base64 encode my data. please convert this issue into a feature request.
Thanks

mlesn, 2018-12-16:
Hello, I have the same need. Is there a chance for implementation? (even as optional code defined in Configuration.hpp.

bblanchon, 2019-02-26:
This feature is available in [version 6.9.0](https://github.com/bblanchon/ArduinoJson/releases/tag/v6.9.0), but it's disabled by default.
To enable it, you need to set [ARDUINOJSON_DECODE_UNICODE](https://arduinojson.org/v6/api/config/decode_unicode/?utm_source=github&utm_medium=issues) to 1 to.

closed 2019-02-26 by bblanchon