bblanchon/ArduinoJson #305: Serial Write Issue

KACIAISSA, 2016-06-15:
Hello, 
I want to ask how to free the buffer after using it, 
I am generating object every 1sec and want to write them to Serial but after 3 or 4 objects 
I get blank object. 
I think it's a buffer issues caused by lake of space. 
any idea how so solve this ?

bblanchon, 2016-06-15:
Hi @KACIAISSA,

I believe you're reusing the `JsonBuffer`, please read:

* [Avoiding pitfalls: Don't reuse the same JsonBuffer](https://github.com/bblanchon/ArduinoJson/wiki/Avoiding%20pitfalls#4-dont-reuse-the-same-jsonbuffer)
* [FAQ: The first serialization succeeds, why do the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-serialization-succeeds-why-do-the-next-ones-fail)
* [FAQ: I found a memory leak in the library!](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#i-found-a-memory-leak-in-the-library)
* [FAQ: What's the best way to use the library?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)

clayfreeman, 2016-06-17:
I've created an issue (#307) that addresses this topic.  Please see if it might help you achieve what you're attempting to do.

closed 2016-06-21 by bblanchon