bblanchon/ArduinoJson #308: How to copy a JsonObject ?

jm4r7in, 2016-06-18:
I would like to do something like that : 
````
void loop() {
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& objectJSON = jsonBuffer.createObject();

  if (stringComplete) {
    // parse string to objectJSON 
  }

  doSomethingWith(objectJSON );

}
````
My problem is that for some reasons, I cannot put the function `doSomethingWith(objectJSON )` in the scope of ` if (stringComplete) {}`... It would be too easy...
Do you have some suggestion ?

Thanks a lot !

bblanchon, 2016-06-18:
Hi Jerome,

I think the problem is that you want a `JsonObject` to have a longer lifetime that it's `JsonBuffer`, which is impossible.
Why don't you extract the relevant information into a custom structure, as shown [here](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)?

jm4r7in, 2016-06-18:
Hi Benoit,

Thanks for your answer !
Yes, it's what I'd like to do. But I'd like this custom structure to be a JSON-like object (because I don't know what will be inside).
Do you know how I can make a JSON-like structure ?

bblanchon, 2016-06-19:
OK, now I understand that what you want is to be able to transport the JSON object trough different part of the app.

There are two ways to accomplish that:

1. you make the `JsonBuffer` global, so you're sure that it will outlive the `JsonObject`.
2. you bundle the `JsonBuffer` and the `JsonObject` in a struct, like below.

```c++
struct JsonBundle {
public:
    JsonBundle(const char* json) {
        _jsonVariant = _jsonBuffer.parseObject(json);
    }

    const JsonObject& root() const { 
        return _jsonVariant;
    }

private:
     DynamicJsonBuffer _jsonBuffer;   
    JsonVariant _jsonVariant;
};
```

Note that I used a `const char*` instead of a `char*` to force a copy of the input string in the JSON buffer.
I also used a `JsonVariant` instead of a `JsonObject&` because you cannot reassign a C++ reference.
Finally, I used a `DynamicJsonBuffer` because the 200 bytes of your code snippet will probably not be enough to store the copy of the input, plus the `JsonObject` hierarchy.

jm4r7in, 2016-06-19:
Thank you Benoit for your answser, but I don't manage to implement it. My knowledge of C++ is rather limited...

1. I tried to make the `JsonBuffer` global, but I can't make it work : 

````
Static JsonBuffer<200> jsonBuffer;

void loop() {
  if (stringComplete) {
    JsonObject& objectJSON = jsonBuffer.parseObject(json);
  }

  JsonObject& objectJSON = jsonBuffer.createObject();
  doSomethingWith(objectJSON );

}
````
How to create a `JsonObject ` from a previously used `JsonBuffer` ?

2. How do you use this structure ?
`struct JsonBundle jsonBundle;` throws an error... :/

bblanchon, 2016-06-20:
Hi Jerome,

Yes, as I said in my previous response, you cannot reassign C++ references.
The trick is to use a  `JsonVariant` that will wrap the reference for you.
(Note that I'll remove references in v6 because many users are facing this issue).

Then, the constructor of `JsonBundle` should not throw an error, can you elaborate?

jm4r7in, 2016-06-20:
When I do : 

````
struct JsonBundle {
  public:
    JsonBundle(const char* json) {
      _jsonVariant = _jsonBuffer.parseObject(json);
    }
  
    const JsonObject& root() const { 
      return _jsonVariant;
    }
  
  private:
    DynamicJsonBuffer _jsonBuffer;   
    JsonVariant _jsonVariant;
};

struct JsonBundle MyJsonBundle;
````

I get : 
````
error: no matching function for call to 'JsonBundle::JsonBundle()
````

How should I use this structure ?

Thanks a lot !

bblanchon, 2016-06-20:
Try:

```c++
struct JsonBundle {
  public:
    void parse(const char* json) {
      _jsonVariant = _jsonBuffer.parseObject(json);
    }

    const JsonObject& root() const { 
      return _jsonVariant;
    }

  private:
    DynamicJsonBuffer _jsonBuffer;   
    JsonVariant _jsonVariant;
};

JsonBundle MyJsonBundle;
```

jm4r7in, 2016-06-20:
Thanks Benoit.
Now, I'm trying to parse a buffer with : 
````
void loop() {
  if (stringComplete) {
    MyJsonBundle.parse(inBuffer);
  }
}
````
But I get : 
````
error: request for member 'parse' in 'MyJsonBundle', which is of non-class type 'JsonBundle()'
````

And to access the `JsonVariant`, `MyJsonBundle.root()` should do it, or am I wrong ?

bblanchon, 2016-06-20:
Did you write:

```c++
JsonBundle MyJsonBundle();
```

instead of

```c++
JsonBundle MyJsonBundle;
```

:question: 

If so, you need to remove the parents.
This is the [most vexing parse](https://en.wikipedia.org/wiki/Most_vexing_parse) and it's terrible for programmers used to C# or Java.

jm4r7in, 2016-06-20:
Thanks Benoit,
`JsonBundle MyJsonBundle;` is working, and to parse, I do (in loop) `MyJsonBundle.parse(inBuffer);`.
It works only twice, and then I get `parseObject() failed`...
The `inBuffer`is filled through serial events and the parsing is done only when the full inBuffer is received.
I really don't get what I'm doing wrong :/

bblanchon, 2016-06-21:
Try:

```c++
struct JsonBundle {
  public:
    void parse(const char* json) {
      _jsonBuffer = DynamicJsonBuffer()
      _jsonVariant = _jsonBuffer.parseObject(json);
    }

    const JsonObject& root() const { 
      return _jsonVariant;
    }

  private:
    DynamicJsonBuffer _jsonBuffer;   
    JsonVariant _jsonVariant;
};

JsonBundle MyJsonBundle;
```

But don't keep a reference to the `JsonObject` between each call of `parse()`, always use `JsonBundle::root()`.

See:

* [FAQ: The first parsing succeeds, why do the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-parsing-succeeds-why-do-the-next-ones-fail)

jm4r7in, 2016-06-21:
`MyJsonBundle::root()` gives the error `'MyJsonBundle' is not a class, namespace, or enumeration` :/

bblanchon, 2016-06-21:
`JsonBundle::root()` is how we name the `root` method of the `JsonBundle` class in C++.
To call this method on an instance of the class, you use the `.` as usual: `MyJsonBundle.root()`.

jm4r7in, 2016-06-21:
According to [FAQ: The first parsing succeeds, why do the next ones fail?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#the-first-parsing-succeeds-why-do-the-next-ones-fail) we shouldn't reuse the `JsonBuffer`..
But it's what we do, with the global `JsonBundle`...

Maybe we could try to make a working example for Arduino. I suggest to start with : 
````
#include <ArduinoJson.h>

#define INBUFFER_SIZE   256
char inBuffer[INBUFFER_SIZE];
bool jsonComplete = false;

struct JsonBundle {
  public:
    void parse(const char* json) {
      _jsonVariant = _jsonBuffer.parseObject(json);
    }

    const JsonObject& root() const { 
      return _jsonVariant;
    }

  private:
    DynamicJsonBuffer _jsonBuffer;
    JsonVariant _jsonVariant;
};

JsonBundle MyJsonBundle;

void doSomething() {
  if (MyJsonBundle.root().containsKey("sn")) {
    // ...
  }
}

void loop() {
  bool parseSuccess = true;
  if ( jsonComplete ) {
    // when json is complete, we parse it, only once
    MyJsonBundle.parse(inBuffer);
    if ( ! MyJsonBundle.root().success() ) {
      parseSuccess = false;
    }
    jsonComplete = false;
  }

  if( parseSuccess ) {
    doSomething();
  }
}

void serialEvent() {
  // ... fill `inBuffer `and set `jsonComplete `as `true `when all data are received
}
````

But it's working only few times...
What do you think about it ?
Again, thanks a LOT for your help :)

bblanchon, 2016-06-21:
Indeed, `JsonBuffer`s are designed to be throwable/one-time-use object.
What I'm showing you is a workaround that makes sure that the `JsonObject` and `JsonBuffer` are both in memory at the same time.
Now, for a simple program as shown here, `JsonBundle` really introduce needless complexity.

Consider this instead:

```c++
// only inBuffer is global, because it's the only thing that needs to persist each loop iteration
char inBuffer[INBUFFER_SIZE];

void doSomething(JsonObject& root) {
  if (root.containsKey("sn")) {
    // ...
  }
}

void loop() {

  readWhatever(inBuffer);

  StaticJsonBuffer<JSON_BUFFER_SIZE> jsonBuffer;
  JsonObject& root = jsonBuffer.parseObject(inBuffer);

  if (root.success()) {
    doSomething(root);
  }
}
```

This seems to fit your requirements.

jm4r7in, 2016-06-21:
It's the first thing I tried, but I have the problem that the parsing should be inside `if ( jsonComplete ) { }` (because I don't want to parse the `inBuffer` if it's not ready) and the `doSomething()` has to be in the loop (and not in the `if ( jsonComplete ) { }` (because this code is generated and put automatically into the loop). See my first question...

bblanchon, 2016-06-23:
> the parsing should be inside if ( jsonComplete ) { }

That should not change anything, just add an `if` statement in `loop()`

>  the doSomething() has to be in the loop (and not in the if ( jsonComplete ) { }

I don't understand, please explain.

jm4r7in, 2016-06-26:
Ok...

Just a very simple question : 
You said : "Why don't you extract the relevant information into a custom structure, as shown [here](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#whats-the-best-way-to-use-the-library)?"

All information is relevant, and I want to store it as a JSON structure.
So instead of using the custom structure `SensorData` as in the example, is it possible to use your library to create a JSON custom structure ?
Or do you know any JSON library that is not designed as a throwable/one-time-use object ?

Thanks a lot.

bblanchon, 2016-06-27:
All of this is a side-effect of the fixed allocation strategy of ArduinoJson.
This is a nearly unique feature that allows ArduinoJson to run on MCU with as little as 1KB of memory.

You can take any other JSON library, and you won't have that particular problem.

Feel free to close this issue if you don't use ArduinoJson anymore.

jm4r7in, 2016-06-27:
I still want to use ArduinoJson but you didn't give any concrete answer...
"Is it possible to use ArduinoJson to create a global JSON custom structure ?"
Thanks for trying anyway...

bblanchon, 2016-06-27:
It's possible to use a global `JsonBuffer` if that's what you meant.

closed 2016-07-05 by bblanchon