bblanchon/ArduinoJson #312: How to determine length of array

retrojdm, 2016-06-23:
Is it possible to get an array size / item count? For example, the "items" array in the example below:
```
{
	"ID": 1234,
	"items": [
		{
			"itemID": "5001":
			"name": "Item One"
		},
		{
			"itemID": "5002":
			"name": "Item Two"
		}
	]
};
```

bblanchon, 2016-06-24:
It's [`myArray.size()`](https://github.com/bblanchon/ArduinoJson/wiki/API%20Reference#jsonarraysize).

I believed this was a well-covered topic in the documentation; I'm quite surprised you didn't find it. Do you have any suggestion to improve the documentation?

retrojdm, 2016-06-24:
Sorry, I'm just having trouble casting root["items"] from my example JSON above as a JsonArray.

There doesn't seem to be a template for the JsonObject::get() function that handles nested JsonArrays or JsonObjects.

Forgive me if I'm thinking about this problem all wrong ;)

bblanchon, 2016-06-24:
No biggie :wink: 
Can I close the issue then?

retrojdm, 2016-06-25:
Yes please.

closed 2016-06-26 by bblanchon