bblanchon/ArduinoJson #315: How to decode the json as mentioned in the message ?

solankigithub, 2016-07-04:
Hi, Thanks for providing such a beautifule library.
However, i am trying to figure out to create the json string as below:
[
{"rpm":"860","speed":"50","odometer":"3400"},
{"rpm":"988","speed":"60","odometer":"3404"},
{"rpm":"876","speed":"55","odometer":"3407"}
]
Can you help me with the code how can i achieve this with your library ?

Thanks

heman4t, 2016-07-05:
Hi @solankigithub, kindly refer to wiki. https://github.com/bblanchon/ArduinoJson/wiki/Decoding%20JSON

solankigithub, 2016-07-05:
Thanks Hemant, but my expectation is little different.
I want to encode it rather decode and my final outcome should be the Json string below:
[
{"rpm":"860","speed":"50","odometer":"3400"},
{"rpm":"988","speed":"60","odometer":"3404"},
{"rpm":"876","speed":"55","odometer":"3407"}
]

bblanchon, 2016-07-05:
Hi @solankigithub,

To be fair with @heman4t, your title says "How to decode the json as mentioned in the message ?"

Here is the documentation you're looking for:
https://github.com/bblanchon/ArduinoJson/wiki/Encoding%20JSON

Don't hesitate to tell us if anything is unclear in the documentation.

solankigithub, 2016-07-05:
Thanks but things are not working as per my expectation.
When i try code :+1:   
```
JsonArray& obdpacket = jsonPacketBuffer.createArray();
JsonObject& obd = obdpacket.createNestedObject("obd");
```
This gives me compiler error: 
`"no matching function for call to 'ArduinoJson::JsonArray::createNestedObject(const char [4])'"`

But when i am trying :
```
  JsonObject& obdpacket = jsonPacketBuffer.createObject();
  JsonObject& obd = obdpacket.createNestedObject("obd");
```
I am getting the result :
`{"obd":{"timestamp":"2016-07-04 14:23:00","rpm":"879","speed":"67","distance":"8455","latitude":"34.9494949","longitude":"98.11"}}
`

I want an array of objects rather an object of object.

Would be great if you can provide the sample code to generate the below json string:
```
[
{"rpm":"860","speed":"50","odometer":"3400"},
{"rpm":"988","speed":"60","odometer":"3404"},
{"rpm":"876","speed":"55","odometer":"3407"}
]

```

bblanchon, 2016-07-05:
Unlike `JsonObject::createNestedObject()`, `JsonArray::createNestedObject()` takes no parameter.
Just remove `"odb"` and you're good to go

solankigithub, 2016-07-06:
Thanks @bblanchon 
Its working absolutely fantastic !
But i've now another problem to resolve which is not related to this api.

I'm facing memory issues since my packet size is exceeding the allocated space. I want to send 15 rows in a packet to the cloud, each row is of 200 characters. so each packet have size 3000 characters, which is equivalent to 3000 bytes or 3Kb. I am using Arduino Uno which has got only 2Kb of Memory space. Send small packet size will be costly as hit to the cloud has cost.

I know this question is not related to this api, but any help would be highly appreciated.

Thanks
Sandeep

bblanchon, 2016-07-06:
You're welcome.

I assume you're using the Arduino Ethernet Shield, which has a 16KB internal memory.
I never tested, but I guess you could fill that buffer and then send everything at once.
As you said, here is not the right place for that kind of question.

On the ArduinoJson side, you'll have to serialize piece by piece (i.e., not all the array at once) because this library requires that all the data sits in memory.

Anyway, this is going to be difficult. My advice is to switch to a bigger hardware or reduce your expectations.

solankigithub, 2016-07-08:
Thanks @bblanchon 
I am using Arduino Uno that has got 2KB of RAM and 32KB of flash memory.
I can not use the 32KB since my data is generated dynamically. We can only use flash memory if the data is static and does not change. 
Your library works real good if i use it alone(i.e. not along with any API) in my program. My compulsion is that i've to use other API's like SIM900.h (GPRS API) and MCPCAN (To read data from CAN). My objective is to read data from CAN convert it into JSON and then send it to the cloud vis GPRS.

I get the error when i call the below method of ArduinoJSON api:
obdpacket.printTo(packet, sizeof(packet));

Does this method occupy too much of memory ?

Below is my Code:
```
#define PACKET_SIZE 200
#define PACKET_REC_COUNT 1
void loop()
{
  char packet[PACKET_SIZE];
  /* OBD Variables */
  int mRPM = 0;
  int mSpeed = 0;
  int mDistance = 0;
  /* GPS Variables */
  
  char lon[15];
  char lat[15];
  char alt[5];
  char time[20];
  char vel[5];

  StaticJsonBuffer<PACKET_SIZE>  jsonBuffer;
  JsonArray& obdpacket = jsonBuffer.createArray();
  int count = 0;
  while (count < PACKET_REC_COUNT)
  {
    count++;
    mRPM=0; mSpeed=0; mDistance=0;
    mRPM = getRPM();
    //Serial.print("RPM : " );
    //Serial.println(mRPM);

    if (mRPM > 600) 
    {
      ignitionStarted = true;   //Check if the Engine is started or Not.
    }
    else{
      ignitionStarted = false;            
    }

    //ignitionStarted = true;
    if (ignitionStarted == true)
    {      
      if (tripCreated == false)
      {
        tripkey = createTrip(getDistance());   //Create New Trip
        tripCreated = true;
        Serial.println(F("Trip Created"));
      }
    

      mSpeed = getSpeed();                    //Get Speed of the Vehicle
      //Serial.print(F("Speed : "));
      //Serial.println(mSpeed);
      mDistance = getDistance();              //Get Odometer Reading of the vehicle
      //Serial.print(F("Odomoter : " ));
      //Serial.println(mDistance);
      gps.getPar(lon, lat, alt, time, vel);   //Get GPS co-ordinates

      JsonObject& obd = obdpacket.createNestedObject();

      obd["t"] = "2016-07-04 14:23:00";
      obd["r"] = mRPM;
      obd["s"] = mSpeed;
      obd["o"] = mDistance;
      obd["lt"] = lat;
      obd["ln"] = lon;

/*      
      obd["t"] = "2016-07-04 14:23:00";
      obd["r"] = "2500";
      obd["s"] = "200";
      obd["o"] = "99999";
      obd["lt"] =   "34.949494999999";
      obd["ln"] =  "98.117777777777";
*/  
    }
    else
    {
      if (tripCreated == true)
      {
        //Update Trip
        updateTrip(tripkey, 0, mDistance);
        Serial.println(F("Trip Updated"));
        tripCreated = false;
        tripkey = 0;
      }
      else
      {
        Serial.println(F("Ignition not started..."));
      }
    } //End If IgnitionStarted

    delay(PER_RECORD_LAG);
    
  } //End While
  obdpacket.printTo(packet, sizeof(packet));
  Serial.println(packet);  
  postOBDData(tripkey, packet); //Post Data on Server
  //Serial.println(F("End While"));  
  
}

```

bblanchon, 2016-07-08:
At a first glance it looks OK.
However, we don't know how much memory are other library using.

You should try obdpacket.printTo(Serial) to see if the problem is related to `packet`.

Also, there is a misconception here: `StaticJsonBuffer<PACKET_SIZE>`.
The template parameter of `StaticJsonBuffer` is not the length of the serialized JSON, it's the size of the object tree.

See:

*  [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)

solankigithub, 2016-07-08:
It works fine when i use
`obdpacket.printTo(Serial)`
But question is, if it works fine with `obdpacket.printTo(Serial)` why it doesn't work for
`obdpacket.printTo(packet, sizeof(packet));`

bblanchon, 2016-07-08:
Most likely because `sizeof(packet)` is too small to hold the serialized string.
How long is the string printed in `Serial`?

solankigithub, 2016-07-08:
It prints forever in the `loop()` without any issues.
I think this is memory issues, when i comment the call of functions, the code starts working.
But the moment i remove the comments , it shows funny output and stops .
Both the `obdpacket.printTo(packet, sizeof(packet));` and `obdpacket.printTo(Serial)` works fine without calling the API functions. The moment i call the api functions in `loop()` it doesn't work.

It seems its a low memory issue. Any workaround you can suggest ?

bblanchon, 2016-07-08:
Yes, it does look like memory issues.

I the code snippet, it sounds like you're using about 500bytes of stack memory.
This looks OK, but I don't know about the rest of your program.

I guess that the other libraries you're using are probably using a lot of memory.
I need to check on this side.

solankigithub, 2016-07-08:
I am thinking of using this library to trace the memory allocation
https://github.com/McNeight/MemoryFree

If you want i can send you the full code on your email. But it will be a pain for you to analyze it.
Please let me know if you want to take that pain. :)

Thanks a ton ! You have been a great support so far !

bblanchon, 2016-07-08:
In the past I used the following function:

```c++
int freeRam () {
  extern int __heap_start, *__brkval; 
  int v; 
  return (int) &v - (__brkval == 0 ? (int) &__heap_start : (int) __brkval); 
}
```

Source: http://playground.arduino.cc/code/AvailableMemory, at the bottom of the page.

You're welcome for the support.
Unfortunately, I don't have time to help you with these memory issues.
This is very specific to your project, only you can find the culprit.

As a general advice: never use `malloc`/`free`, nor `new`/`delete`, nor `String` nor any form of heap memory in an Arduino project.
On this topic see: https://blog.benoitblanchon.fr/arduino-json-parser/

Also, a GitHub star is always appreciated :wink:

closed 2016-07-08 by bblanchon

solankigithub, 2016-07-08:
Thanks ! I will figure out this but upgrading my hardware will be the last option.. I think i still have scope to improve my code.

Many Thanks ! 5 star to you ! :+1:

bblanchon, 2016-07-09:
Note: I added [JsonArray::createNestedArray()](https://github.com/bblanchon/ArduinoJson/wiki/API-Reference#jsonarraycreatenestedobject) to the documentation.