bblanchon/ArduinoJson #317: how to pass a JsonArray& to a function

mckennatim, 2016-07-08:
give an input`{"id": 0, "pro":[[0,0,0,84,64],[6,30,1,84,70]]}`

I would like to have my function that decodes the input to be able to call a function with a signature something like this 
`void copyProg(prg_t& t, JsonArray& ev)`

I can hardcode it without a function call and it works fine. Once I have `id` I'd like to copy the `pro` array into id=0's data structure. 

    void Sched::deseriProg(prgs_t& prgs, char* kstr){
      StaticJsonBuffer<300> jsonBuffer;
      JsonObject& rot = jsonBuffer.parseObject(kstr);
      int id = rot["id"];
      JsonArray& events = rot["pro"];
      for(int h=0;h<events.size();h++){
        JsonArray& aprg = events[h];
        aprg.printTo(Serial);
        for(int j=0;j<prgs.temp1.numdata+2;j++){
          prgs.temp1.prg[h][j] = aprg[j];
          Serial.print(prgs.temp1.prg[h][j]);
        }
      }

But when I try to pull the loop into a function

    void copyProg(prg_t& t, JsonArray& ev){
      for(int h=0;h<ev.size();h++){
        JsonArray& aprg = ev[h];
        for(int j=0;j<t.numdata+2;j++){
          t.prg[h][j] = aprg[j];
        }
      }        
    }

and call it from here

    void Sched::deseriProg(prgs_t& prgs, char* kstr){
      StaticJsonBuffer<300> jsonBuffer;
      JsonObject& rot = jsonBuffer.parseObject(kstr);
      int id = rot["id"];
      JsonArray& events = rot["pro"];
      switch(id){
       case 0:
         copyProg(prgs.temp1, events);          
         break;
       case 1:
         copyProg(prgs.temp2, events);          
         break;
       ...
       default:
          Serial.println("in default");
      }     

I get errors compiling with 1.67 on esp8266 Wemos D1 mini

    sketch\Sched.cpp.o: In function `Sched::printSched(int)':
    sketch/Sched.cpp:502: undefined reference to `Sched::copyProg(prg_t&, ArduinoJson::JsonArray&)'

    sketch\Sched.cpp.o: In function `get<int>':

    sketch/Sched.cpp:502: undefined reference to `Sched::copyProg(prg_t&, ArduinoJson::JsonArray&)'

    collect2.exe: error: ld returned 1 exit status

I am sure this is my issue in not understanding `JsonArray&` but still I'd love some help.

psigen, 2016-07-08:
It looks like you are having a linking error which is probably unrelated to `JsonArray` itself.  The line:
```
sketch/Sched.cpp:502: undefined reference to `Sched::copyProg(prg_t&, ArduinoJson::JsonArray&)'
```
indicates that the compiler thinks that you should have declared a function with the signature:
`Sched::copyProg(prg_t&, ArduinoJson::JsonArray&)`
but it was unable to find it in your code.  There could be many reasons for this, for example:
1) Your `copyProg` function was included in the `class Sched` declaration, but was defined as a standalone function.
2) Your `copyProg` function has a slightly different signature than the one that you called (perhaps one of the types is slightly mismatched)

mckennatim, 2016-07-10:
@psigen, You are right. I pulled the code out my big app and it compiles and runs. Hope I find the simple error somewhere and have not hit upon some limit for arduino 1..67 /esp8266 in linking many files. I have plenty of storage space and global variable space.

closed 2016-07-10 by mckennatim

mckennatim, 2016-07-10:
@psigen, found it. `void copyProg(prg_t& t, JsonArray& ev)` needed to be `void Sched::copyProg(prg_t& t, JsonArray& ev)`