bblanchon/ArduinoJson #321: Need help json pointers

flyingbutter, 2016-07-13:
Hİ,
I parsed 16 json strings but i need to put them in a loop.I tried to store their adresses as following;
JsonObject* info[16]={&root1,&root2,&root3,&root4,&root5,&root6,&root7,&root8,&root9,&root10,&root11,&root12,&root13,&root14,&root15,&root16};

When i try to use these values compiler gives the error: invalid types 'ArduinoJson::JsonObject*[const char [5]]' for array subscript
roots are all jsonobjects. 
ı would try to put all these objects in a json array but ı have no idea how it can be done.Can someone give me an idea how to use these in a loop?
Thanks.

bblanchon, 2016-07-13:
Hi,
I don't know why you're having this error, but if you need to store `JsonObject`s in a `JsonArray`, you can do that:

```c++
JsonArray& array = jsonBuffer.createArray();

for (int i=0; i<16; i++) {
   JsonObject& obj = jsonBuffer.parseObject(/* insert JSON here */);
   array.add(obj);
}
```

But be carefull with memory consumption!

flyingbutter, 2016-07-13:
thanks for the answer but ı am using an esp8266 with around 82kb max memory but i kind of found a solution. I store json strings in a 2 dimonsional array and just parse it and use it in loop and it also saved me a lot of memory.

closed 2016-07-13 by flyingbutter

bblanchon, 2016-07-13:
You're welcome @flyingbutter, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink: