bblanchon/ArduinoJson #322: Is it possible to use containsKey method for nested keys?

gabrielebellini, 2016-07-14:
Hello,
having this json with nested object:
{"key1":5,"body":{"key2":7,"key3":9}}
Is there a way to use containsKey for finding nested keys?
root.containsKey("key2") return false because it's inside "body" key. Could this function be use to traverse all levels and find, for example, "key2" or "key3" presence?

Thanks in advance.

Gabriele

bblanchon, 2016-07-14:
Hi Gabriele,

No. There is no function to do that.

However, you can implement it yourself, like this:

```c++
bool containsNestedKey(const JsonObject& obj, const char* key) {
    for (const JsonPair& pair : obj) {
        if (!strcmp(pair.key, key))
            return true;
      
        if (containsNestedKey(pair.value.as<JsonObject>(), key)) 
            return true;
    }

    return false;
}
```

Please add this to [Wiki: Bag of Tricks](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks) if it works.

Regards,
Benoit.

gabrielebellini, 2016-07-14:
Thank you Benoit for your help! It works flawlessly. Your JSON library is a masterpiece! Very useful and elegant. As requested, I just added your suggested solution to the Bag of Tricks in the Wiki. Thank you again! Gabriele

bblanchon, 2016-07-15:
You're welcome, Gabriele.
Thank you for using ArduinoJson and for you contribution on the Wiki.

One last thing, you can show your support by adding a star to the project on GitHub; this is always appreciated :wink:

closed 2016-07-15 by bblanchon

gabrielebellini, 2016-07-15:
Added one star 👍 Thanks again for your precious work.

enurseitov, 2016-08-16:
Arduino IDE's compiler gives error: ```
error: range-based 'for' loops are not allowed in C++98 mode
     for (const JsonPair& pair : obj) {
                                 ^
```

bblanchon, 2016-08-16:
@enurseitov Newer version of Arduino IDE use C++11.

If for some reason, you cannot upgrade, here is the C++98 version:

```c++
bool containsNestedKey(const JsonObject& obj, const char* key) {
    for (JsonObject::const_iterator it=obj.begin(); it!=obj.end(); ++it) {
        if (!strcmp(it->key, key))
            return true;

        if (containsNestedKey(it->value.as<JsonObject>(), key)) 
            return true;
    }

    return false;
}
```

enurseitov, 2016-08-16:
@bblanchon That's strange, I use rather new Arduino IDE v. 1.6.6. Anyway, thanks a lot for fast reply, I'll try your code later.

wizbowes, 2017-04-22:
Hi - I'm trying this currently but get the following compile error:
`undefined reference to `ArduinoJson::JsonObject ArduinoJson::JsonVariant::as<ArduinoJson::JsonObject>() const'`

I'm calling the function like this:

`if (containsNestedKey(root,"speed")) {Serial.println("yay");} else {Serial.println("nah");}`

Not sure if that's correct - pointers confuse me.