bblanchon/ArduinoJson #323: Parse JsonObject in JsonArray

dkijkuit, 2016-07-15:
I have the following JSON strting (which I cannot modify because it's from an external system):
`{"sensors":[{"sensor_id":"18fe34cd040c","temperature":23,"humidity":31,"timestamp":1468526592562,"sensor_status":"ONLINE"},{"sensor_id":"18fe34cedab6","temperature":26,"humidity":34,"timestamp":1468526589786,"sensor_status":"ONLINE"}]}`

I parse the JSON as follows:
```
    char* msg_payload = (char*)payload;
    DynamicJsonBuffer jsonBuffer;
    JsonObject& root = jsonBuffer.parseObject(msg_payload);
```

The parsing succeeds and now I want to get the array inside:
`JsonArray& nestedArray = root["sensors"].asArray();`

How do I get the array elements as JsonObject? I've tried several things, like this
    `JsonObject& jsonObj = nestedArray.get(0);`

...  but that doesn't work either. Can you help me out? Thanks in advance!

bblanchon, 2016-07-15:
Hello David,

There are two syntaxes, depending on your position with regards to implicit conversions.

The first syntax uses implicit conversion from `JsonVariant` to `JsonObject&`

```c++
JsonObject& obj = array[0];
```

The second syntax requires a template argument:

```c++
JsonObject& obj = array.get<JsonObject&>(0);
```

However, it's also possible to omit the template argument, this defaults to `JsonVariant`.

So the code you've written should work.
Please try the other syntaxes and tell me what happens.

dkijkuit, 2016-07-16:
You're right, this works as expected:
```
JsonArray& nestedArray = root["sensors"].asArray();
JsonObject& jsonObj = nestedArray.get<JsonObject&>(0);
const char* sensor_id = jsonObj["sensor_id"];
                
USE_SERIAL.println(sensor_id);
```

Thank you for this great library and your assistance!

bblanchon, 2016-07-16:
You're welcome David, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-07-16 by bblanchon