bblanchon/ArduinoJson #324: root.printTo( buff, sizeof(buff)) introduces : at random times

jescarri, 2016-07-20:
code:

```
char *build_json( boolean dht_read_ok, float t, float h, float hic, double pressure, int dht_failures){
  char buff[JBUFF];
  char p_buff[15];
  StaticJsonBuffer<200> jsonBuffer;
  JsonObject& root = jsonBuffer.createObject();
  root["node_id"] = "S1";
  JsonArray& data = root.createNestedArray("sensor_list");
  data.add("DHT");
  #if defined (BMPPRESENT)
    data.add("BMP");
  #endif
  if ( dht_read_ok == true ){
    root["satus"] = "ok";
    root["vcc"] = ESP.getVcc();
    root["dht_failures"] = dht_failures;
    root["temp"] = t;
    root["humidity"] = h;
    Serial.print("Humidity at json conv ");
    Serial.println(h);
    root["hic"] = hic;
    #if defined (BMPPRESENT)
       root["pressure"] = atof(deblank(dtostrf(pressure,11,3,p_buff)));
    #endif
    root.printTo(buff, sizeof(buff));
  }else{
    root["satus"] = "error";
    root["dht_failures"] = dht_failures;
    root.printTo( buff, sizeof(buff));
  }
  return buff;
}
```

Latest Arduino Json Library

Output:

```
.......
WiFi connected
IP address: 
192.168.1.39
18:FE:34:CF:7D:0F
SeaLevel Comp pressure 
1004.64
nan
nan
Failed to read from DHT sensor!
44.00
27.40
 Reconecting
Attempting MQTT connection...connected
Humidity at json conv 44.00
Temp at json conv 27.40
HIC at json conv 27.40
{"node_id":"S1","sensor_list":["DHT","BMP"],"satus":"ok","vcc":2770,"dht_failures":1,"temp":27.3:,"humidity":44.00,"hic":27.40,"pressure":1.00e3}
Entering deep sleep mode for 1 secs 
```

Notice the : at the humidity reading

a good json output:

```
...........
WiFi connected
IP address: 
192.168.1.39
18:FE:34:CF:7D:0F
SeaLevel Comp pressure 
1004.67
nan
nan
Failed to read from DHT sensor!
44.00
27.50
 Reconecting
Attempting MQTT connection...connected
Humidity at json conv 44.00
Temp at json conv 27.50
HIC at json conv 27.49
{"node_id":"S1","sensor_list":["DHT","BMP"],"satus":"ok","vcc":2770,"dht_failures":1,"temp":27.50,"humidity":44.00,"hic":27.49,"pressure":1.00e3}
Entering deep sleep mode for 1 secs 
```

jescarri, 2016-07-20:
Parse errors:
```
795: unexpected token at '{"node_id":"S1","sensor_list":["DHT","BMP"],"satus":"ok","vcc":2769,"dht_failures":1,"temp":27.3:,"humidity":44.40,"hic":27.43,"pressure":1.00e3}'
795: unexpected token at '{"node_id":"S1","sensor_list":["DHT","BMP"],"satus":"ok","vcc":2767,"dht_failures":1,"temp":27.3:,"humidity":44.2:,"hic":27.42,"pressure":1.00e3}'
795: unexpected token at '{"node_id":"S1","sensor_list":["DHT","BMP"],"satus":"ok","vcc":2770,"dht_failures":1,"temp":27.3:,"humidity":44.2:,"hic":27.42,"pressure":1.00e3}'
```

It looks that is something regarding the float conversion

bblanchon, 2016-07-20:
This bug is fixed in version 5.6.4.

@jescarri Thank you very much for reporting this issue :+1:

closed 2016-07-20 by bblanchon

jescarri, 2016-07-20:
Thanks for the quick turnaround!!! 
Appreciate it!.