bblanchon/ArduinoJson #327: prettyPrintTo in example JsonGeneratorExample

zixzixzix, 2016-08-01:
` root.prettyPrintTo(Serial);`
this print `{
"sensor": "gps",
"time": 1,
"data": [
4.756080,
2.302038`
but root.printTo(Serial);
print`{"sensor":"gps","time":1351824120,"data":[48.756080,2.302038]}
`

bblanchon, 2016-08-01:
Here is a screen capture, I took this morning with Arduino IDE 1.6.10 and a Dueminalove:

[image]

There must be something special in your setup.

Can you provide more context?

zixzixzix, 2016-08-01:
[image]
[image]

bblanchon, 2016-08-01:
It seems to be a problem with line-endings.

At first, I believed that this was because you are running on OSX.
So I installed Arduino on my Mac, but I don't have the same problem.

The issue must be somehow related to the mkr1000.
Please make sure you're using the latest core for SAMD (it seems to be 1.6.6).

If the problem persists, please try to run this:

```c++
Serial.println("line1");
Serial.println("line2");
Serial.print("line3");
Serial.println();
Serial.print("line4");
Serial.println();
Serial.print("line5");
```

And the output should be:

```
line1
line2
line3
line4
line5
```

zixzixzix, 2016-08-02:
I  try this code and it work fine.
And the issue  is the same  on windows.

bblanchon, 2016-08-02:
I ordered a mkr1000; hopefully, I'll manage to understand what going on here.
It's coming from the UK so I'll only have it next week.

bblanchon, 2016-08-09:
I just received the board.
I'll test this ASAP.

bblanchon, 2016-08-10:
OK, I managed to reproduce the bug on mkr1000.
Give me some more time to figure out what going on.

zixzixzix, 2016-08-10:
Thank u

bblanchon, 2016-08-11:
That was not easy, but I found it!!!!

It was a bug in Arduino SAMD Core.
It's been fixed on May 27th by https://github.com/arduino/ArduinoCore-samd/commit/8ef9bb3d7bd026b57a8568030d8992644c9b8de8 and https://github.com/arduino/ArduinoCore-samd/commit/8d0c1674628df1c2c7592f4fc17467c694f5a1be.
Unfortunately, version 1.6.6 only includes modifications up to May 19th, so the fix is not available through the Arduino Boards Manager.

I recommend to apply the modification manually or wait until the next release of the Arduino SAMD Core.

closed 2016-08-23 by bblanchon