bblanchon/ArduinoJson #329: Parsing fails when a json contains more than 9 key:value pairs.

shtarbanov, 2016-08-06:
My static json buffer is set to 1000. 

Initially I thought that the reason for failure had to do with the string length, but after performing the following tests, I noticed that the string length was irrelevant. What mattered was the number of key:value pairs. Here are the tests I conducted:
10 keys, 821 characters long -----> FAILS
10 keys, 804 characters long -----> FAILS
10 keys, 724 characters long -----> FAILS
  9 keys, 755 characters long -----> WORKS.
  9 keys, 819 characters long -----> WORKS.

Here is an example of a json string that fails to parse:
{"dev1":{"state":true,"power":82,"delay_sec":5,"config":false,"yyyy":"xxxxxxxxxx"},"b":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"c":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"d":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"e":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"f":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"g":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"h":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"i":{"state":true,"power":82,"delay_sec":300,"config":false,"yyyy":"xxxxxxxxxx"},"j":{"state":true,"power":19,"delay_sec":5,"config":false,"yyyy":"xxxxxxxxxx"}}

bblanchon, 2016-08-07:
It looks like the `JsonBuffer` is running out of space.
That is probably because it has to make a copy of the input.
Make sure you use a `char[]`.

See:

* [FAQ: Why parsing fails?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#why-parsing-fails)
* [API Reference: `JsonBuffer::parseObject()`](https://github.com/bblanchon/ArduinoJson/wiki/API-Reference#jsonbufferparseobject)

shtarbanov, 2016-08-07:
It is neither of those 2. I have checked.
I have satisfied my needs with an alternative implementation that uses only one key:value pair per json string, but the problem aforementioned is still worth investigating.

bblanchon, 2016-08-07:
Please post a small program that reproduces the issue and specify the target platform.

BTW, are you sure that you have enough RAM to store the input and the JsonBuffer.
For instance, an Arduino has only 2KB of RAM; it's not enough for 1000 + 724 bytes, because you need extra spaces for the rest of the variables.

See also:

* [ArduinoJson: Memory model](https://github.com/bblanchon/ArduinoJson/wiki/Memory%20model)
* [FAQ: How to determine the buffer size?](https://github.com/bblanchon/ArduinoJson/wiki/FAQ#how-to-determine-the-buffer-size)

bblanchon, 2016-08-15:
@shtarbanov Did you find a solution? Can I close this issue?

closed 2016-08-23 by bblanchon