bblanchon/ArduinoJson #330: JsonVariant has no test for "key":null

gerikintercede, 2016-08-11:
There is no way to test for "key":null in a JsonVariant. 
JsonVariant().is<char*>() returns false as expected, so is no use.
JsonVariant().asString() returns NULL, but it will also return NULL for "key":1, so is no use.

Needs something like:
```
inline bool JsonVariant::isNull() const {
  using namespace Internals;
  return (_type == JSON_UNPARSED && _content.asString &&
      !strcmp("null", _content.asString));
}
```
However, this also needs **bool JsonVariant::is< void* >();**, or possibly **class JsonNull; bool JsonVariant::is< JsonNull >();**

Also there are no tests in JsonVariant_As_Tests.cpp and JsonVariant_Is_Tests.cpp for "key":null .

bblanchon, 2016-08-12:
When you write `variant = (const char*)0`, `is<char*>()` returns `true`, I think that's what most people expect.
So I think it would be better if `is<char*>()` returned `true` when the input contains a `null`.
In your case, you'd just have to write:

```c++
bool isNull(const JsonVariant& v) {
  return v.is<char*>() && !v.as<char*>();
}
```

`is<void*>()` won't do it because most implementations define `NULL` as an `int`.

The `JsonNull` will be confusing. Look:

```c++
char * s = 0;
JsonVariant v = s;
s.is<char*>();  // false or true ?
s.is<JsonNull>(); // true: it feels like the type has changed
```

In C++11, we could also leverage `nullptr` and `nullptr_t`, but I want to keep the library C++98 compliant for now.

gerikintercede, 2016-08-12:
Not so.
A variant with from "key":null results in a JSON_UNPARSED type with _content.asString = “null”
Therefore `is<char*>()` returns `false`.
```
  template <typename T>
  typename TypeTraits::EnableIf<TypeTraits::IsSame<T, const char *>::value ||
                                    TypeTraits::IsSame<T, char *>::value,
                                bool>::type
  is() const {
    return isString();
  }
  bool isString() const {
    return _type == Internals::JSON_STRING;
  }
```

Perhaps you need to modify `isString()` to match the code in `asString()`:
```
  bool isString() const {
    if (_type == JSON_UNPARSED && _content.asString &&
        !strcmp("null", _content.asString))
      return true;
    return _type == Internals::JSON_STRING;
  }
```

bblanchon, 2016-08-12:
Sorry, my message was unclear.

When I said "it would be better if `is<char*>()` returned true", I meant that I see this as a bug in `is<char*>()`.

I'll fix this ASAP.

bblanchon, 2016-08-15:
Fixed in v5.6.5
Sorry for the long delay.

closed 2016-08-15 by bblanchon

milkpirate, 2018-01-01:
Sorry for my question but I really couldnt figure out what to do to check if `"key":null` is null.

So will
```
bool isNull(const JsonVariant& v) {
  return v.is<char*>() && !v.as<char*>();
}

if( isNull(root["key"]) ) {
   // yes, key == null 
}
```
work?

bblanchon, 2018-01-02:
Hi @milkpirate,

Yes, this `isNull()` function would work, but it's probably possible to avoid this check.
Why do you need to know if the value is `null`?

Regards,
Benoit

milkpirate, 2018-01-02:
I transmit settings from a web form via a JSON string as URL parameter (its easier to parse it that way instead of every setting by its own). If not all settings are changed I set them to `null` and discard them in further processing.