bblanchon/ArduinoJson #331: Constructing JSON array of objects iteratively in main loop()

JonSilver, 2016-08-14:
I have an ESP8266 application which receives sensor inputs in bursts, datum by datum as it cycles around the loop(). Each burst of data is isolated using a timeout, and then processed to perform a calculation, the result of which is then published via MQTT. For testing and data modelling purposes, I'd also like to be able to send the raw data for each burst via a single JSON message. Thus I'd need to build the JSON object iteratively, adding each received datum to an array, and once the timeout occurs publish the JSON object, and discard and recreate the jsonbuffer.

I want to do this in a best-practice way rather than using workarounds to clear the jsonbuffer. Please could someone give me any clues as to the best way of achieving this?

If I declare the jsonbuffer etc globally, I'd need to clear it at some point => BAD
If I declare the jsonbuffer etc locally, they go out of scope after each iteration => BAD
If I declare the jsonbuffer etc as static, they stay in scope through iterations, but still need clearing => BAD

Is the answer to temporarily stash/copy the contents of the locally declared jsonbuffer and associated objects in a static char[] before they go out of scope? Or am I just wildly guessing?

Apologies if that has been answered before, but I searched for hours and couldn't find my way. I'm still learning huge amounts every day, so all help gratefully received. Thanks in advance.

Partial code for illustration, with no attempt at the ArduinoJSON bits:

```
void loop() {
   if (newData) {
      timeout = 0;
      // stash the data in a json array
   }

   if (timeout > 250) {
      timeout = 0;
      // send the json, start again from new
   }
}
```

bblanchon, 2016-08-15:
I would probably use a vector/array of struct to store the temporary data and then serialize everything in a separate function.

```c++
static std::vector<MyDatum> data;

void loop() {
   if (newData) {
      timeout = 0;
      data.push_back(/*...*/);
   }

   if (timeout > 250) {
      timeout = 0;
      sendData(data);
      data.clear();
   }
}
```

Or I would go with the global object route, possibly wrapping everything in a class that will send and clear within the same function.

As a general rule, I recommend not to use a JsonArray as storage because there is a significant overhead.

Then I avoid mixing serialization and business rules in the same part of the code.

JonSilver, 2016-08-15:
Thank you so much, Benoît... very good advice (and very quick response). I shall try that, and I think at the same time I may try restructuring things within my polling loop().

bblanchon, 2016-08-15:
You're welcome Jon.
Thank you for using ArduinoJson.

closed 2016-08-15 by bblanchon

JonSilver, 2016-08-15:
Just one more thank you Benoît - I have a perfectly working firmware now, and have learned much in the process.

bblanchon, 2016-08-15:
Great! I'm glad I could help 😄