bblanchon/ArduinoJson #332: How to add a JSON object in another JSON object

jeferrb, 2016-08-15:
I have a set of JSON objects and I want to append one in the other, but I have any idea of how to do that. Spousing that I have:

```
char json1[] = R"({
    "fingerprint": "1234"
  })";

char json2[] = R"({
    "Trait1":{
        "someValue": "4321"
    }
  })";

char json3[] = R"({
    "Trait2":{
        "anotherValue": "5555"
    }
  })";

JsonObject& jsonObjectRoot = jsonBuffer.parseObject(json1);
JsonObject& jsonObjectOne  = jsonBuffer.parseObject(json2);
JsonObject& jsonObjectTwo  = jsonBuffer.parseObject(json3);
```
How can I add the **jsonObjectOne** and the **jsonObjectTwo** inside the **jsonObjectRoot** to get a JSON that looks like:

```
{
    "fingerprint": "1234",
    "SetOfObjects" : {
        "Trait1":{
            "someValue": "4321"
        },
        "Trait2":{
            "anotherValue": "5555"
    }
  }
}
```

So many thanks.

bblanchon, 2016-08-16:
Try something like this (not tested):

```c++
void merge(JsonObject& dest, JsonObject& src) {
   for (auto kvp : src) {
     dest[kvp.key] = kvp.value;
   }
}

JsonObject& nestedObject = jsonObjectRoot.createNestedObject("SetOfObjects");
merge(nestedObject, jsonObjectOne);
merge(nestedObject, jsonObjectTwo);
```

If you find this function useful, please add it to [Wiki: Bag of Tricks](https://github.com/bblanchon/ArduinoJson/wiki/Bag%20of%20Tricks).

jeferrb, 2016-08-16:
Works like expected.
Thanks so much Benoît.
That is exactly what I wanted.

bblanchon, 2016-08-16:
You're welcome Jeferson, thanks for using ArduinoJson.

Don't hesitate to edit the wiki if you think you can improve it.
In particular, feel free to add your projects to [Projects using ArduinoJson](https://github.com/bblanchon/ArduinoJson/wiki/Projects%20using%20ArduinoJson).

Also, a GitHub star is always appreciated :wink:

closed 2016-08-16 by bblanchon